ARMmbed/mbed-os #2993: [LWIP] - Wrong header inclusion

pan-, 2016-10-12:
### Description
- Type: Bug |

The file [`emav_lwip.c`](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/emac_lwip.c) include the header `platform.h` which includes `C++` standard headers. Therefore, the compilation fail on GCC because standard C++ headers are not accessible from C files.

This bug can be hidden by the duplicate `platform.h` in the code base, depending on the project, the compiler can pick the one in `mbedtls` or the one in `platform`.

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
0.9.5

**meed-os sha:**
a551faa881512432b54200c4c57a41cafcd0af02

pan-, 2016-10-12:
@0xc0170 What could we do about this ? This might cause compilation errors for every projects using lwip.

0xc0170, 2016-10-12:
The emac file referenced above should use config file to get device macros (used to be ``device.h``).

I'll have a look and send a propose a fix.

pan-, 2016-10-12:
what about the duplication of `platform.h` ? 
Should I open an issue for this and if so, is it and mbed tls issue ?

0xc0170, 2016-10-12:
>  Should I open an issue for this and if so, is it and mbed tls issue ? 

Yes, lets create a separate issue. Is not already in the issue tracker - having duplicated header files in the code base ?

MarceloSalazar, 2016-10-14:
Fixed in https://github.com/ARMmbed/mbed-os/pull/2994

closed 2016-10-14 by MarceloSalazar