ARMmbed/mbed-os #2995: Exporter Project.py error: No module named xdg.BaseDirectory

bcostm, 2016-10-12:
### Description
Type: Bug
Related issue: none
Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F103RB

**Toolchain:**
ARM uVision5

**mbed version:** 
0.9.1

**mbed sha:**
(`git log -n1 --oneline`)
f5fb485 Merge pull request #2967 from viquiram/master

**Expected behavior**
zip file containing uVision 5 project

**Actual behavior**
```
Traceback (most recent call last):
  File "..\mbed-os\tools\project.py", line 14, in <module>
    from tools.export import EXPORTERS, mcu_ide_matrix
  File "C:\GITHUB\mbed-os\tools\export\__init__.py", line 18, in <module>
    from tools.export import codered, ds5_5, iar, makefile
  File "C:\GITHUB\mbed-os\tools\export\iar\__init__.py", line 12, in <module>
    from tools.export.cmsis import DeviceCMSIS
  File "C:\GITHUB\mbed-os\tools\export\cmsis\__init__.py", line 9, in <module>
    from xdg.BaseDirectory import save_data_path
ImportError: No module named xdg.BaseDirectory
```

**Steps to reproduce**
Launch the command:
`mbed-os\tools\project.py -c -m NUCLEO_F103RB -p 0 -i uvision5`

0xc0170, 2016-10-12:
cc @theotherjimmy @sarahmarshy

theotherjimmy, 2016-10-12:
`pip install -Ur requirements.txt`

bcostm, 2016-10-12:
OK thanks, no more error now. I just have this warning:
 ```
C:\Python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this wa
rning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
```

I tried to install the python-Levenshtein module but it failed...

Anyway, the generated zip file is not correct. A lot of source files and folders are missing in it.

JojoS62, 2016-10-12:
> I tried to install the python-Levenshtein module but it failed...
I got also an error, the log said that a VC90 compiler is required for compilation some c sources (on windows). 

> Anyway,  the generated zip file is not correct. A lot of source files and folders are missing in it.
The same that I reported also already:  #2882 
It looks like the the tools/project.py does not work without CLI anymore, is this true and intended? In older mbed_os versions the export structure was the same as in the lib-source. I've tried to add the sources as before but the files will not be placed in the right subdirectories.

theotherjimmy, 2016-10-13:
@JojoS62 @bcostm could you provide singer concrete examples of the behavior you expect vs the behavior you see?

JojoS62, 2016-10-13:
[MBED_BLINKY_uvision_LPC1768.zip](https://github.com/ARMmbed/mbed-os/files/527921/MBED_BLINKY_uvision_LPC1768.zip)
[MBED_BLINKY_lpcxpresso_LPC1768.zip](https://github.com/ARMmbed/mbed-os/files/527922/MBED_BLINKY_lpcxpresso_LPC1768.zip)
Ok, hope the upload works. 
I have attached two projects generated with the older mbed-master version of 2015/12. One is for uVision, one for lpcxpresso.
Both work as desired: the uVision must be unzipped to a dest folder, then the project can be started with the .uvproj file.
For lpcxpresso, you start the IDE and use the import provide the dialog with the generated .zip file. Then the files are copied from the zip to the eclipse workspace directory.
Most IDE's that I know (eclipse based) use the directory structure that you see in the lpcx project file. This is also documented in the actual docs/building.md, starting at line 511.
The uVision uses a different layout and not all files are visible.
I have also attached screenshots of how it looks like in these IDE's.
[image]
[image]
Now with the actual mbed-os build all of the mbed path is missing, directory structure is not build as before. This is due to the restructuring of the target/hal/cmsis stuff. Before it was easier to create because it was a copy the source structure, now some kind of remapping is necessary.
I tried also the project export in mbed-cli, but there are only the .project files generated. The ide does not pull only the necessary files from this project file, the import grabs the whole target stuff and tries to compile 2mio files...

theotherjimmy, 2016-10-18:
> Before it was easier to create because it was a copy the source structure, now some kind of remapping is necessary.

What you have noticed is that tests do not have the same structure as a normal project. This is intended behavior.

>The ide does not pull only the necessary files from this project file, the import grabs the whole target stuff and tries to compile 2mio files...

That's actually an issue.

This issue is getting far from it's original topic name. Could we change it, or open another issue?

bcostm, 2016-10-18:
You're right. I have renamed it for clarity.

JojoS62, 2016-10-22:
The export handling is now different than with mbed-master: the mbed-cli project contains everything for all targets and features. Project files for IDE's are generated in the root of the project and the IDE should use now mbed-cli directory structure. Am I right?
For some IDE's its working already, but with the given .project for LPCXpresso the IDE scans all directories and tries to compile everything.
This can be avoided by setting 'resource filters' in the project (should be true for other eclipse based IDE's also). Then the same workflow as for example uvision or gcc_arm is possible.
I have imported the sample project 'mbed-os-example-blinky' and created the export for lpcxpresso. Then I started the LPCXpresso and imported the blinky, unchecking the 'copy to workspace' to use the original location.
Then I added resources filters in the project options and the folders that contain subfolders that need also filtering. In the attachement you will find the modified .project file with a new 'filtered-resources' section. These entries must be generated by the exporter of course. Each filter contains an id, which looks like a simple unique number, does someone know what it meas?
Another way may be to use virutal folders (linked resources), but I haven't tried this yet.
Hope these experiements help to get the lpcxpresso 
[lpcxpresso-projectfiles.zip](https://github.com/ARMmbed/mbed-os/files/545701/lpcxpresso-projectfiles.zip)
export working.

theotherjimmy, 2016-11-02:
> Project files for IDE's are generated in the root of the project and the IDE should use now mbed-cli directory structure. Am I right?

Yes

Thanks for doing some of the research @JojoS62 ! This will really help get the lpcxpresso exporter working quicker.

I'm curious about another thing withe regard to the filtered resources section: what do those arguments mean.

Further, I would like to know if we can just specify all of the source to include up front.

JojoS62, 2016-11-03:
>

JojoS62, 2016-11-03:
> I'm curious about another thing with regard to the filtered resources section: what do those arguments mean?

> Further,
 I would like to know if we can just specify all of the source to include up front.

I thing the filtered resources are the key to what you want. This means I can select 'include everything in this path' or 'include only this'. Don't know if is really handy, but I understand you want a same solution for all exporters. I'm busy in some projects, hope I find some time this weekend to get this working.
When the exporter is called, do I get the scanned result (necessary directory structure) for building the filtered resources tree?

theotherjimmy, 2016-11-03:
> When the exporter is called, do I get the scanned result (necessary directory structure) for building the filtered resources tree?

Yes, in a Resources object that is the resources member of the Exporter class. Here are some code snippets that you might find useful:

 * The attributes of the Resources object: https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/__init__.py#L47-L77
 * which files go where: https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/__init__.py#L606-L658
 * an example of using the resources object in an exporter: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/makefile/__init__.py#L45-L61

sg-, 2017-01-19:
Is this resolved?

0xc0170, 2017-04-26:
> Is this resolved?

I dont think so. I exported via ``python tools\project.py -m LPC1768 -i uvision -p 83`` from the latest master, and the project is incorrect as reported above. all files have incorrect paths (the workspace assumes the files are inside the exported directory, that is not correct).

One of the errors:
```
"no source": Error:  #5: cannot open source input file "main.cpp": No such file or directory
```

The project zipped actually has files with correct paths but does not build neither (this zipped file I assume is the one exported in the online IDE), some files are missing. I wonder why there is this difference a project exported in export folder vs the one inside there in .zip file.

One of the errors:

```
TARGET_NXP\TARGET_LPC176X\serial_api.c(17): error:  #5: cannot open source input file "mbed_assert.h": No such file or directory
```