ARMmbed/mbed-os #2997: Docs: Building is outdated (requires update to new structure + compilers, etc)

0xc0170, 2016-10-12:
### Description
- Type: Bug 
- Priority:  Minor

## Bug

Docs https://github.com/ARMmbed/mbed-os/blob/master/docs/BUILDING.md is outdated. It contains information about mbed-sdk.

I would propose to rename it to ```BUILDING_MBED2.md`` and update it to the new tree structure + compilers and other relevant information that has changed. Any other ideas?

@sg- @bridadan @c1728p9 

Thanks @pan- for pinpointing this.

sg-, 2016-10-12:
Should be completely re-written using mbed-cli as the interface. Need to align this with the update of mbed 2 test and mbed 5 unification.

bridadan, 2017-01-17:
This file has been removed as of https://github.com/ARMmbed/mbed-os/pull/3475. Should we be working adding another doc here, adding a doc to the handbook repo, or should this issue be closed?

closed 2017-01-17 by sg-