ARMmbed/mbed-os #2998: [mbed tls|platform] Duplicate header: platform.h

pan-, 2016-10-12:
### Description
- Type: Bug 

The header `platform.h` is available twice in the code base: [here](https://github.com/ARMmbed/mbed-os/blob/master/platform/platform.h) and [there](https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/inc/mbedtls/platform.h).

This might lead to an incorrect header inclusion and compilation errors.

pan-, 2016-10-12:
@0xc0170 @sg- What can we do about this ? 
I've notice this from the OOB testing, in one project, `platform.h` from `platform` is picked up and this lead to an error in the LWIP module (reported [here](https://github.com/ARMmbed/mbed-os/issues/2993#issuecomment-253214214)). 
For other projects or for mbed-os tests, it is the header from mbed-tls which is used for compiling `emac_lwip.c`. 
This might be the case for other modules as well.

sg-, 2016-10-12:
What do the api and design guidelines say?

pan-, 2016-10-12:
@sg- ooops forgot about this one. 
I can do the list of files which don't follow that rule with `platform.h`. 
Should I open an issue for each module ?

sg-, 2016-10-12:
that works!

sg-, 2017-01-19:
Looking at our platform.h it seems we should just delete it and move those system includes into mbed.h and same with the pin type headers.

closed 2017-10-27 by ChiefBureaucraticOfficer