ARMmbed/mbed-os #3000: [MTS_MDOT_F405RG] Blinky does not compile: USBTX/USBRX not defined

bulislaw, 2016-10-12:
### Description
- Type: Bug
- Priority: Minor

**Target**
MTS_MDOT_F405RG

**Toolchain:**
GCC_ARM

**meed-os sha:**
master (f5fb485dcd4e6ed8d913d61da453ae04b804ec11)

'''
$  mbed compile -t GCC_ARM -m MTS_MDOT_F405RG
Building project mbed-os-example-blinky (MTS_MDOT_F405RG, GCC_ARM)
Scan: ..
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Compile [  0.4%]: AnalogIn.cpp
Compile [  0.8%]: BusIn.cpp
Compile [  1.2%]: BusInOut.cpp
Compile [  1.6%]: BusOut.cpp
Compile [  2.0%]: CAN.cpp
Compile [  2.3%]: Ethernet.cpp
Compile [  2.7%]: main.cpp
Compile [  3.1%]: FileLike.cpp
Compile [  3.5%]: FileBase.cpp
Compile [  3.9%]: FilePath.cpp
Compile [  4.3%]: I2CSlave.cpp
Compile [  4.7%]: I2C.cpp
Compile [  5.1%]: FileSystemLike.cpp
Compile [  5.5%]: LocalFileSystem.cpp
Compile [  5.9%]: InterruptIn.cpp
Compile [  6.2%]: InterruptManager.cpp
Compile [  6.6%]: SPISlave.cpp
Compile [  7.0%]: RawSerial.cpp
Compile [  7.4%]: SPI.cpp
Compile [  7.8%]: Stream.cpp
Compile [  8.2%]: Serial.cpp
Compile [  8.6%]: SerialBase.cpp
Compile [  9.0%]: Timer.cpp
Compile [  9.4%]: Timeout.cpp
Compile [  9.8%]: Ticker.cpp
Compile [ 10.2%]: TimerEvent.cpp
Compile [ 10.5%]: equeue_posix.c
Compile [ 10.9%]: equeue.c
Compile [ 11.3%]: greentea_serial.cpp
[Error] greentea_serial.cpp@5,52: 'USBTX' was not declared in this scope
[Error] greentea_serial.cpp@5,59: 'USBRX' was not declared in this scope
[ERROR] ../mbed-os/features/frameworks/greentea-client/source/greentea_serial.cpp: In constructor 'GreenteaSerial::GreenteaSerial()':
../mbed-os/features/frameworks/greentea-client/source/greentea_serial.cpp:5:52: error: 'USBTX' was not declared in this scope
 GreenteaSerial::GreenteaSerial() : mbed::RawSerial(USBTX, USBRX) {};
                                                    ^
../mbed-os/features/frameworks/greentea-client/source/greentea_serial.cpp:5:59: error: 'USBRX' was not declared in this scope
 GreenteaSerial::GreenteaSerial() : mbed::RawSerial(USBTX, USBRX) {};
                                                           ^

[mbed] ERROR: "python" returned error code 1.
```

bridadan, 2016-10-12:
@bulislaw That target isn't an mbed OS 5 target. But it should still have those two pins defined, even for mbed 2!

bulislaw, 2016-10-12:
Ah right the other Mdot is: MTS_MDOT_F411RE my bad. @bridadan feel free to close this issue.

closed 2016-10-12 by bridadan

bridadan, 2016-10-12:
Also super important: This was the 3000th PR/Issue!