ARMmbed/mbed-os #3002: [IAR IDE] Debugger setting incorrect for IAR targets

bridadan, 2016-10-12:
### Description
- Type: Bug/Enhancement
- Priority: Minor

---------------------------------------------------------------
## Bug/Enhancement

**Target**
Tested with K64F, NRF51_DK, and NUCLEO_F411RE

**Toolchain:**
IAR

**Toolchain version:**
IAR IDE version 7.40

**mbed-cli version:** 
0.9.7

**meed-os sha:**
Tested with oob blinky example hash 7ec525eb49cfdf9942eb5b5caaec2306d216d17c, which uses mbed-os hash of a551faa881512432b54200c4c57a41cafcd0af02

**Expected behavior**
Hitting "Download and Debug" should download the built project to target and start a debug session

**Actual behavior**
All targets default to the debug settings "Simulator", which does not program the target

**Steps to reproduce**
`mbed export -m <TARGET> -i iar`
Build the project, then click "Download and Debug"

---------------------------------------------------------------
## Further Notes

I did some investigation on the K64F, NRF51_DK, and NUCLEO_F411RE. I attempted to find the correct debug settings for these targets. I was able to find the correct settings for the NUCLEO_F411RE. I got the K64F and the NRF51_DK attempting to flash, however they ended up failing. I've attached the relevant project files and debug logs here: [iar_debug_settings.zip](https://github.com/ARMmbed/mbed-os/files/525335/iar_debug_settings.zip)

cc @theotherjimmy @sarahmarshy

sarahmarshy, 2016-10-12:
Can you verify this is a problem with 7.5? I was not able to replicate this on 7.5. Given that I developed debugging files for 7.5, this is likely the issue.

bridadan, 2016-10-13:
After switching to 7.70, I was able to get the K64F flashing once I change the debugger from `Simulator` to `CMSIS-DAP` and the protocol from `JTAG` to `SWD`. But I still wan't able to flash the NRF51_DK.

And I don't have a download link for 7.50 (the oldest version on the IAR website is 7.60) so I can't test that unfortunately.

0xc0170, 2016-10-13:
> After switching to 7.70, I was able to get the K64F flashing once I change the debugger from Simulator to CMSIS-DAP and the protocol from JTAG to SWD

This should be set in the template, thus still something wrong.

@sarahmarshy Not certain, but I havent spotted any version numbers for our exporters (mainly templates, what versions are supported). It might be somewhere that I missed it?

I am using IAR 7.70 and uvision 5.17, as those versions have been used for releases for last couple of months.

sarahmarshy, 2016-10-13:
Interesting. I was able to step through blinky on a k64f last night, so IAR version difference seems to be the culprit. I used 7.5. For uvision, I used 5.21a.

Should we come to a consensus about a version and move forward? The logical choice seems like the latest releases of each, as we cant maintain support for all versions. I'm not really free to look into this until Monday, but if we agree to use the latest version, can you investigate the necessary template changes for IAR debugging?

sarahmarshy, 2016-10-13:
The best approach is to generate the project files, open them, and save the project (IAR does some xml rearranging on a save), then move them to a backup location. Then, you can do the same, but change the debugging settings and save. After that, you can diff the respective ewp and ewd files. Keep note of the xml tags, and find them in the .ewd and .ewp templates in the IAR folder of export.

bogdanm, 2016-10-13:
@0xc0170 @sarahmarshy: if I understood correctly, the version should be 7.70. I tried to reproduce this issue locally with both 7.10 and 7.70, and it's reproducible with both versions. Debugging works, but I have to set `CMSIS DAP` in `Debugger/Simulator` and `SWD` in `Debugger/CMSIS DAP`. Those should be the defaults.

bogdanm, 2016-10-13:
@0xc0170: please confirm the IAR versions that we need to support, so we can move forward with this.

sarahmarshy, 2016-10-13:
They are the defaults. I was definitely able to debug with 7.5 last night. Very frustrating that the templates are that different. I'm throwing in my vote for 7.7.

bogdanm, 2016-10-13:
Sorry, I meant "the default for the IAR version that we decide to support with this release".

0xc0170, 2016-10-13:
@sg- confirmation for v7.70 to be supported for IAR and uvision to be v5.21?

sg-, 2016-10-13:
Always the latest version 👍 for exporters

bridadan, 2016-10-13:
I'm fine with 7.70 moving forward, I'm in the process of upgrading the CI machines to use that at the moment.

0xc0170, 2016-10-20:
Should be resolved by https://github.com/ARMmbed/mbed-os/pull/3070

closed 2016-10-20 by 0xc0170