ARMmbed/mbed-os #3007: Blinky online compilation fails for K64F with 5.2.0-rc2

infinnovation, 2016-10-13:
### Description
- Type: Bug
- Related issue: `#2961`
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64

**Toolchain:**
Whatever the online IDE uses

**meed-os sha:**
a551faa

**Expected behavior**
Compilation would succeed after updating mbed-os to latest revision.
Fault would have been detected & fixed after rc1.

**Actual behavior**
Two errors:
```
Error: "/tmp/glm2vL", line 120 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
Error: Undefined symbol mbedtls_hardware_poll (referred from entropy.K64F.o).
```

**Steps to reproduce**
New program -> Template: mbed OS Blinky LED HelloWorld
In Program Workspace, right-click on mbed-os -> Update...
The Library Details pane now shows Revision 2541:a551faa
Compile

0xc0170, 2016-10-13:
cc @screamerbg and @adbridge , you both have noticed the same correct.

We could not reproduce this offline, as ``mbedtls_hardware_poll`` should not be required (mbedtls was updated to use TRNG HAL). Some misconfiguration

adbridge, 2016-10-13:
I compiled offline for the following combinations with no issues:
1) mbed-os-example-blinky, updated with tag mbed-os-5.2.0-rc2, compiled with GCC_ARM for K64F
2) oob-mbed-os-example-blinky, updated with tag mbed-os-5.2.0-rc2, compiled with GCC_ARM for K64F

Only thing I noticed was this warning: 
Object file test_env.o is not unique! It could be made from: c:\users\annbri01\work\mbed-os-example-blinky\mbed-os\features/unsupported\tests\mbed\env\test_env.cpp .\mbed-os\features\frameworks\greentea-client\source\tes
t_env.cpp

infinnovation, 2016-10-13:
It works for me too offline with GCC_ARM - that's not the problem.  Does it work with "-t ARM"?  Is it something to do with the online IDE environment?

infinnovation, 2016-10-18:
The problem may be as much with switching between versions as with the particular version of mbed-os.  Two procedures to demonstrate the problem follow.  Firstly new-old-new:

Import mbed-os-example-blinky (we now get mbed-os 2246:a6f3fd1)
Compile - OK
Switch mbed-os to 2243:d220204
Compile - Errors
```
Error: Undefined symbol lp_ticker_clear_interrupt (referred from mbed_lp_ticker_api.K64F.o).
Error: Undefined symbol lp_ticker_disable_interrupt (referred from mbed_lp_ticker_api.K64F.o).
Error: Undefined symbol lp_ticker_init (referred from mbed_lp_ticker_api.K64F.o).
Error: Undefined symbol lp_ticker_read (referred from mbed_lp_ticker_api.K64F.o).
Error: Undefined symbol lp_ticker_set_interrupt (referred from mbed_lp_ticker_api.K64F.o).
```
Update mbed-os (back to 2264:a6f3fd1)
Compile - Errors
```
Error: "/tmp/AcykwU", line 120 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
Error: Symbol greentea_serial multiply defined (by test_env.K64F.o and greentea_serial.K64F.o).
Error: Not enough information to list the image map.
```

By contrast, if the first compilation is against the older mbed-o (old-new-old):

Import mbed-os-example-blinky (get mbed-os 2246:a6f3fd1)
Switch mbed-os to 2243:d220204
Compile - OK
Update mbed-os to 2246:a6f3fd1
Compile - Errors
```
Error: "/tmp/VrZXh1", line 120 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
Error: Symbol greentea_serial multiply defined (by test_env.K64F.o and greentea_serial.K64F.o).
Error: Not enough information to list the image map.
```
Switch mbed-os back to 2243:d220204
Compile - OK

@screamerbg Maybe something is not cleaned up when switching versions?

theotherjimmy, 2017-04-26:
@immunda I'll look into this one and track it with the Core stuff.

ChiefBureaucraticOfficer, 2017-09-25:
Unable to reproduce with mbed OS 5.5

closed 2018-05-31 by deepikabhavnani