ARMmbed/mbed-os #3012: Export keil project failed in OS 5.2

cyliangtw, 2016-10-13:
Export keil project failed in OS 5.2, even I update project-generator as (0.9.10) & project-generator-definitions as (0.2.33).
It seems the Keil environment MDK 5.17 can't provide existing arm-pack-manager data path.

```
$  mbed export -i uvision5 -m K64F
............................
[ ERROR ] file http://www.keil.com/pack/index.idx did not download (28, 'Operation too slow. Less than 51200 bytes/sec transferred the last 2 seconds')
Caching Files 1/0

### cpu_name=MK64FN1M0xxx12
### cpu_name=ARMCM4_FP
Traceback (most recent call last):
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\project.py", line 234, in <module>
    main()
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\project.py", line 230, in main
    zip_proj=zip_proj, build_profile=profile)
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\project_api.py", line 229, in export_project
    macros=macros)
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\project_api.py", line 90, in generate_project_files
    exporter.generate()
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\export\uvision\__init__.py", line 192, in generate
    'device': DeviceUvision(self.target),
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\export\uvision\__init__.py", line 23, in __init__
    DeviceCMSIS.__init__(self, target)
  File "C:\ARM_mbed\examples\eth_tcp_Hello\mbed-os\tools\export\cmsis\__init__.py", line 58, in __init__
    raise TargetNotSupportedException("Target not in CMSIS packs")
tools.export.exporters.TargetNotSupportedException: Target not in CMSIS packs
```

0xc0170, 2016-10-13:
cc @theotherjimmy @sarahmarshy 

Exporters do not depend on project_generator anymore, mbed tools contain now starting with 5.2 own exporters. Reasons you can find in the pull request that pulled in those changes.

The traceback here that a device is not in cmsis_pack, however I haven't seen any issue for k64f. We will look into it.

Probably we should state somewhere the state of exporters (what they depend on, how they work, how to add new targets, where to find them, etc).

bridadan, 2016-10-13:
@theotherjimmy Looks like the arm_pack_manager tool failed to download the cmsis pack?

@cyliangtw I believe the first time you run the exporters in 5.2, you need an internet connection so the mbed tools can download the necessary info. Did you have a network connection at the time?

theotherjimmy, 2016-10-13:
@bridadan yes that's what or looks like.

cyliangtw, 2016-10-14:
As usual, I only enable HTTPS connection proxy for pip install & git operation in my environment.
Thanks, it could work after I also enable HTTP connection proxy.

closed 2016-10-14 by cyliangtw