ARMmbed/mbed-os #3018: NUCLEO_F401RE I2C broken

c1728p9, 2016-10-13:
### Description
- Type: Bug
- Priority: Major

## I2C failures with EEPROM (I2C)
I2C -  EEProm Read (fails with ARMCC, GCC)
I2C -  TMP102 Temperature Read (fails with GCC, works with ARMCC)

**Target**
NUCLEO_F401RE

**Toolchain:**
GCC_ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC: @adustm @bcostm @screamerbg

c1728p9, 2016-10-13:
The commit 8982ff197126caae361c00fbbbd10e833690169d (#2622) appears to have introduced this problem. At the end of a read a nack is no longer sent which causes the IIC device to keep holding SDA low.

Logic capture on 22ec73bc3ab77dbc9e8f361addfb9abca2c430f8 when running test:
[image]

Logic capture on same commit but with 8982ff197126caae361c00fbbbd10e833690169d reverted:
[image]

CC: @egostm

c1728p9, 2016-10-13:
This is likely the cause of the I@C failures in #2956 and #2959.

betzw, 2016-10-14:
Should already be fixed by PR https://github.com/ARMmbed/mbed-os/pull/2972

screamerbg, 2016-10-14:
As @betzw pointed, no longer an issue after PR #2972  @c1728p9

c1728p9, 2016-10-14:
Tested #2972 and it passes the I2C tests. Logic capture also looks correct.

closed 2016-10-14 by c1728p9