ARMmbed/mbed-os #3020: How to port mbed-os for F411CE

nightseas, 2016-10-14:
### Description
- Type: Question
- Target: STM32F411CE

---------------------------------------------------------------
## Question

I'm trying to port mbed-os and classic mbed lib to STM32F411CE.
The problem is the startup assembly file. Even if I replaced the file with the one my cube-mx project for 411CE, it still couldn't run to main().

Any suggestion?

BTW: The HAL lib for STM32 in mbed is really old. I even can't find this package version for cube-mx. Do you have any plan to upgrade it?

Thanks.
BR//Xiaohai Li

nightseas, 2016-10-14:
It works now. Guess it's configuration and GCC compatible issue.

closed 2016-10-14 by nightseas

FilipeChagasDev, 2020-02-11:
Do you have this port? Could you share the target code?

nightseas, 2020-02-16:
Try official mbed-os 5.9 or later version. Export the code for NUCLEO_F411RE to uVision5 by mbed-cli and change target device to F411CE.
It should work.