ARMmbed/mbed-os #3025: export NCS36510 to IAR fails

maclobdell, 2016-10-14:
### Description
- Type: Bug 
- Related issue: n/a
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
IAR IDE

**Toolchain version:**
N/A
**mbed-cli version:** 
tested 0.9.7 & 0.9.9

**mbed-os sha:**
a42457d Merge pull request #3015 from andresag01/update_mbedtls_rc

**DAPLink version:**
n/a

**Expected behavior**
successful export of Iar project

**Actual behavior**
Crash

Traceback (most recent call last):
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project.py", line 234, in <module>
    main()
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project.py", line 230, in main
    zip_proj=zip_proj, build_profile=profile)
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project_api.py", line 229, in export_project
    macros=macros)
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.check_supported()
  File "C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\export\exporters.py", line 126, in check_supported
    raise TargetNotSupportedException()
tools.export.exporters.TargetNotSupportedException
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\maclob01\Documents\mbed-os-5\mbed-os\tools\project.py -i iar -m NCS36510 --source ." in "C:\Users\maclob01\Documents\mbed-os-5\mbed-os"
---

**Steps to reproduce**

mbed export -m NCS36510 -i IAR

maclobdell, 2016-10-14:
I see that the exporter is not using progen and the support for NCS36510 did not make it into the new exporter infrastructure.  I will make a pull request asap.

theotherjimmy, 2016-10-20:
Looks like were good!

closed 2016-10-20 by theotherjimmy