ARMmbed/mbed-os #3029: Nuvoton NUMAKER_PFM_NUC472

screamerbg, 2016-10-14:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Pwm failures (PwmOut) - GCC and ARMCC
Error while trying to initialize PwmOut on D3, D5, D6, D9 (as per Arduino header standard)

Tests available here https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/Pwm/Pwm.cpp
See below for schematics.

## I2C failures with EEPROM (I2C)
I2C -  EEProm Write - 2bytes (but 1 byte words)
I2C -  EEProm Write - 10bytes
I2C -  EEProm Write - 100bytes

Tests available here https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/I2C/I2C.cpp
See below for schematics.

## Analog In failures (AnalogIn) - ARMCC
Analog Input on A0
Analog Input on A1
Analog Input on A2
Analog Input on A3
Analog Input on A4
Analog Input on A5

Tests available here https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/AnalogIn/AnalogIn.cpp
See below for schematics.

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
ARM

**meed-os sha:**
08ff689 Merge pull request #2979 from adustm/STM_F429_F439

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components and schematics available here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

Note that tests are based on Arduino header connectivity standard. More information available here: 

CC @ccli8

cyliangtw, 2016-10-14:
In PFM_NUC472, D9 can't be regarded as Pwm pin, the right pin should be one of D2 ~ D7, the pin diagram of PFM_NUC472 is at: (https://www.mbed.com/en/platform/hardware/boards/nuvoton-technology/nuvoton_nuc472/ )
Could we assign in PinNames.h file or override in mbed_app.json for MBED_CONF_APP_PWM_x ?

cyliangtw, 2016-12-12:
It's fixed by #3309 ( ` https://github.com/ARMmbed/mbed-os/pull/3309` )
So, it could be closed.

bridadan, 2016-12-12:
Thanks @cyliangtw! @screamerbg please reopen if you find the issues haven't been resolved

closed 2016-12-12 by bridadan