ARMmbed/mbed-os #3031: -D options for TARGET_XXX and TOOLCHAIN_XXX not passed to .s files

j3270, 2016-10-14:
### -D options for TARGET_XXX and TOOLCHAIN_XXX not passed to .s files
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
MAX32600MBED

**Toolchain:**
ARM

**Toolchain version:**
MDK-ARM Plus 5.21a

**mbed-cli version:** 
0.9.5

**Expected behavior**
build of single .s file written with support for all 3 toolchains through preprocessor statements

**Actual behavior**
.s file fails to build due to lack of TARGET_ and TOOLCHAIN_ -D option

**Steps to reproduce**
write single asm file that supports all three toolchain; GCC, IAR, ARM. and try to build.

**My Comments**
I have attached to build outputs that use the tools stated above.

1.  Manually provides -D options on cli for target and toolchain, success.
2.  Doesn't provide -D target options for target and toolchain, failure.

The owlink.s file used to build just fine with the online tool for the last 3-4 mos.  I tried to build a project using the library a few weeks ago and it failed due to the assembly code not being built.

The errors are listed below.
[DEBUG] Errors: Error: L6218E: Undefined symbol ow_bit (referred from OwGpio.o).
[DEBUG] Errors: Error: L6218E: Undefined symbol ow_usdelay (referred from OwGpio.o).

I have troubleshot the problem this far, however, am not familiar enough with the development of the tools to troubleshoot it any further.
 
What changed in the (tools) build process? 

[build_with_options.txt](https://github.com/ARMmbed/mbed-os/files/530272/build_with_options.txt)
[build_without_options.txt](https://github.com/ARMmbed/mbed-os/files/530274/build_without_options.txt)

screamerbg, 2016-10-14:
@j3270 This is the intended behavior. There are many macros that are generated by the target, libraries or the top application. Not all of these macros are compatible with all toolchains and some of them have limitations, e.g. length of asm parameters or macro type (for example IAR does not accept float and string macros to iarasm). For compatibility reasons only the macros defined by the target are being passed to asm. Workaround for this is to define target_override in your top application and add/remove macros. If the macro is important to have at all times then add it to targets.json

j3270, 2016-10-14:
Thanks for your response,

I am not sure I understand.  If this is the intended operation, why was the library working up until a few weeks ago.

Did this intended operation just recently become the intended operation?

Thanks,

j3270, 2016-10-14:
@screamerbg please see above

screamerbg, 2016-10-14:
@j3270 then it might be unrelated issue as the limitation to macros was applied during the mbed OS 5.1 release in August.

Perhaps @0xc0170 @geky @theotherjimmy could provide more light on this.

j3270, 2016-10-14:
Thanks,

I think I understand your solution.  The macros could be added in targets.json by the vendor, correct?

j3270, 2016-10-14:
@screamerbg 

I tried adding the macros to the name"macros" under the target object "MAX32600MBED" in targets.json and in latest_targets.json under the .tools folder.  No luck, any suggestions?

Thanks,

j3270, 2016-10-14:
@screamerbg 

One note, these macros were listed in the build output on the line [DEBUG] Macros: at the beginning of the build before any changes were made, they just are not passed to the preprocessor when preprocessing the .s file.  Apparently that is the new intended operation with mbed-os 5.1 that was released in August, which matches the time frame that I started having trouble with the builds.

Additionally, from the build output it appears that .s files are preprocessed before they are assembled/compiled.  Wouldn't this be the place for any incompatible macros to be removed, dealt with?

Thanks

kgills, 2016-10-17:
We can work around the Target definitions by adding them to targets.json, but there are still some syntax differences between the different assemblers. We're still not going to get the TOOLCHAIN_ macros.

bridadan, 2017-01-17:
@j3270 It sounds like the issue raised here is actually the intended behavior. Would you confirm this?

sg-, 2017-01-19:
Yes - this is the intended behavior and it was introduced around mid 5.1.x when exporters refactoring and unification happened. In general macros generated by a scan arent a good thing to use. I'd like to see that enhanced at some point but is low priority now

closed 2017-01-19 by sg-