ARMmbed/mbed-os #3034: Question: mbed export -S "not enough space" error

RobMeades, 2016-10-15:
Running `mbed export -S` on the current `mbed-os` version (#7bcc63b), on Windows 7, `mbed-cli` version 0.9.9, gives me a "not enough space" error:

```
C:\projects\mbed-os-app>mbed export -S
Traceback (most recent call last):
  File "C:\projects\mbed-os-app\mbed-os\tools\project.py", line 234, in <module>
    main()
  File "C:\projects\mbed-os-app\mbed-os\tools\project.py", line 187, in main
    print mcu_ide_matrix()
IOError: [Errno 12] Not enough space
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\projects\mbed-os-app\mbed-os\tools\project.py -S" in "C:\projects\mbed-os-app"
```

What could I be doing to cause this?

0xc0170, 2016-10-17:
@theotherjimmy @screamerbg

bogdanm, 2016-10-17:
See also https://github.com/ARMmbed/oob-mbed-os-example-blinky/issues/26

bridadan, 2016-10-17:
I did a little research and it looks like we may be hitting the maximum string that can be printed at once by Windows in a command prompt? So we may need to break up the string a bit before printing it, or examine how we can reduce the amount of data that's printed.

closed 2016-10-19 by sg-