ARMmbed/mbed-os #3037: `mbed update`does not update libraries if branch is checked out.

SeppoTakalo, 2016-10-17:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**mbed-cli version:** 0.9.7
**meed-os sha:** a551faa

**Expected behavior**

Command `mbed update` should update my `mbed-os` repository to point to the git hash from `mbed-os.lib`

**Actual behavior**
```
$ mbed update
[mbed] Updating program "mbed-os-example-mesh-minimal" to latest revision in the current branch
merge: origin/rc3 - not something we can merge
[mbed] Updating library "atmel-rf-driver" to rev #c1b7ba3c2e9a
[mbed] Updating library "mbed-os" to rev #873686ce791c
[mbed] Updating library "mcr20a-rf-driver" to rev #d8810e105d7d
$ cd mbed-os
$ git log --oneline | head -1
a551faa (HEAD -> mbed-os-5.2, tag: mbed-os-5.2.0-rc2) Merge pull request #2975 from ARMmbed/release-candidate
```

**Steps to reproduce**

1. Git Clone `oob-mbed-os-example-mesh-minimal`
2. `git checkout -b rc2 mbed-os-5.2.0-rc2`
3. `mbed deploy`
5. `cd mbed-os`
4. `git checkout -b mbed-os-5.2 mbed-os-5.2.0-rc2`
5. `git branch --set-upstream-to=origin/mbed-os-5.2`
6. `cd ..`
7. `git checkout -b rc3 mbed-os-5.2.0-rc3`
8. `mbed update`

Now you can verify that `mbed-os` did not update.
Steps 4&5 emulate the state that you have checked out the release branch when RC2 was the head.

0xc0170, 2016-10-17:
@SeppoTakalo Is this mbed-cli issue?

@theotherjimmy  @screamerbg - please have a look

SeppoTakalo, 2016-10-17:
How do I know?
Is is a problem on where the cloning and updating functionality is.

MarceloSalazar, 2016-10-17:
I've followed these steps and work ok to me:

1. `mbed import oob-mbed-os-example-mesh-minimal`
2. `cd oob-mbed-os-example-mesh-minimal`
3. `mbed update mbed-os-5.2.0-rc3`
4. `mbed ls -a`

```
oob-mbed-os-example-mesh-minimal (https://github.com/ARMmbed/oob-mbed-os-example-mesh-minimal#57574347e016)  --> rc3
|- atmel-rf-driver (https://github.com/ARMmbed/atmel-rf-driver#c1b7ba3c2e9a)
|- mbed-os (https://github.com/ARMmbed/mbed-os#873686ce791c)  --> rc3
`- mcr20a-rf-driver (https://github.com/ARMmbed/mcr20a-rf-driver#d8810e105d7d)
```

maybe mbed CLI does not like messing around with git ?

SeppoTakalo, 2016-10-17:
I already know that the workflow you proposed works.
Point is that in cases, where you have manually updated your `mbed-os` to release branch, the `mbed update` command does not work anymore, even that it claims to.

screamerbg, 2016-10-17:
@SeppoTakalo Can you please run `mbed update -v` and also `mbed update -vv` and attach the 2 logs?

screamerbg, 2016-10-17:
Actually nevermind. I'm able to reproduce it

theotherjimmy, 2016-10-20:
@screamerbg  is this resolved in mbed-cli v 0.9.10?

screamerbg, 2016-10-21:
Yes, now fixed.

closed 2016-10-21 by screamerbg