ARMmbed/mbed-os #3038: NCS36510 - lp ticker timeout failure

0xc0170, 2016-10-17:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510 

**Toolchain:**
GCC_ARM, ARM, IAR

**mbed-os sha:**
mbed-os RC3

**Actual behavior**

ncs36510 - tests-mbed_drivers-lp_timeout fails : 

```
01:17:24 [1476512242.40][CONN][RXD] >>> Running case #1: '500us LowPowerTimeout'...
01:17:24 [1476512242.45][CONN][INF] found KV pair in stream: {{__testcase_start;500us LowPowerTimeout}}, queued...
01:17:24 [1476512242.51][CONN][RXD] :86::FAIL: Values Not Within Delta 600 Expected 500 Was 1465
01:17:24 [1476512242.56][CONN][INF] found KV pair in stream: {{__testcase_finish;500us LowPowerTimeout;0;1}}, queued...
01:17:24 [1476512242.65][CONN][RXD] >>> '500us LowPowerTimeout': 0 passed, 1 failed with reason 'Test Cases Failed'
```

@pradeep-gr - please have a look at this failure

pradeep-gr, 2016-10-18:
This test case is passing with PR https://github.com/ARMmbed/mbed-os/pull/2866

pradeep-gr, 2016-10-18:
[image]

pradeep-gr, 2016-10-18:
Attached snap shot to make sure that we are talking about same test.

bridadan, 2016-10-18:
@pradeep-gr When we build and execute tests in CI, we also turn on heap and stack metrics. Perhaps this is affecting something else and causing the issue? Can run the following commands from the latest master and post the log?

```
mbed test --compile -m NCS36510 -t GCC_ARM -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 --clean
mbed test --run -m NCS36510 -t GCC_ARM -n mbed-os-tests-mbed_hal-lp_ticker -v
```

@c1728p9 Do you think that the heap and stack statistics could be affecting this test?

c1728p9, 2016-10-18:
I doubt they would have an effect on the LP ticker

c1728p9, 2016-10-18:
Hi @pradeep-gr, the screenshot you have is for the lp ticker test which is working correctly. The failure occurs on the lp timeout test. You should be able to recreate the problem using this command ```mbed test -t GCC_ARM -m NCS36510 -n tests-mbed_drivers-lp_timeout```

pradeep-gr, 2016-10-24:
Issue resolved with commit 60e9e9e91da600372ce94d256c33240688882488

closed 2016-12-20 by theotherjimmy