ARMmbed/mbed-os #3039: mbed export and project.py sw4stm32 don't work

adustm, 2016-10-17:
### Description
- Type: Bug 
- Priority: Major
---------------------------------------------------------------
## Bug
**Refering to**
https://developer.mbed.org/questions/74000/For-information-about-exporting-to-SW4ST/
**Target**
NUCLEO_F410RE | DISCO_F746 | NUCLEO_F746ZG...
**Toolchain:**
sw4stm32

**Toolchain version:**

**mbed-cli version:** 0.8.2

**meed-os sha:**a42457d

**Expected behavior**
A project is generated 

**Actual behavior**
While investigating the issue https://developer.mbed.org/questions/74000/For-information-about-exporting-to-SW4ST/
I've noticed that the command line exporter does not work for any platforms on the sw4stm32 toolchain. It seems that the project.py cannot recover the 'core' information...
Are you aware of that ?
```
Traceback (most recent call last):
  File "C:\mbed\tools\project.py", line 234, in <module>
    main()
  File "C:\mbed\tools\project.py", line 230, in main
    zip_proj=zip_proj, build_profile=profile)
  File "C:\mbed\tools\project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "C:\mbed\tools\project_api.py", line 229, in export_project
    macros=macros)
  File "C:\mbed\tools\project_api.py", line 90, in generate_project_files
    exporter.generate()
  File "C:\mbed\tools\export\sw4stm32.py", line 82, in generate
    core = self.target.core
AttributeError: 'str' object has no attribute 'core'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\mbed\tools\project.py -i sw4stm32 -m NUCLEO_F401RE --source ." in "C:\MCDGitW601\mbed"
```
**Steps to reproduce**
mbed export -i SW4STM32 -m NUCLEO_F401RE
or
python tools/project.py -m NUCLEO_F401RE -i sw4stm32 -p 90
Both commands give the same error.

0xc0170, 2016-10-17:
cc @theotherjimmy @sarahmarshy

theotherjimmy, 2016-10-18:
@0xc0170 See the linked PR.

adustm, 2016-10-18:
Sorry I did not find it myself. Thanks !
I have checked that your modification unblocks the script, but I don't see the generated project yet.
I'll have to check that later.
@bcostm

theotherjimmy, 2016-10-18:
If you did an mbed export then it's in the root

theotherjimmy, 2016-10-20:
@0xc0170 @adustm This should be resolved as #3060 has been merged.

theotherjimmy, 2016-10-20:
please re-open if this problem persists

closed 2016-10-20 by theotherjimmy