ARMmbed/mbed-os #3043: mbed-os cache not working properly

hasnainvirk, 2016-10-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 

hasnain@ubuntu ~/mbed-os/oob-mbed-os-example-client (mbed-client-rc=)
$ mbed deploy
[mbed] Adding library "esp8266-driver" from "ssh://git@github.com/ARMmbed/esp8266-driver.git" at rev #dc37b65ca877
[mbed] Adding library "atmel-rf-driver" from "ssh://git@github.com/armmbed/atmel-rf-driver.git" at rev #62c0c49d7a94
[mbed] Adding library "mcr20a-rf-driver" from "ssh://git@github.com/ARMmbed/mcr20a-rf-driver.git" at rev #d8810e105d7d
**[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #873686ce791c**
[mbed] Adding library "pal" from "ssh://git@github.com/ARMmbed/pal.git" at rev #08f7c93f8e24
[mbed] Adding library "mbed-client" from "ssh://git@github.com/ARMmbed/mbed-client.git" at rev #1c03b36ac944
[mbed] Adding library "mbed-client/mbed-client-mbed-tls" from "ssh://git@github.com/ARMmbed/mbed-client-mbed-tls.git" at rev #7e1b6d815038
[mbed] Adding library "mbed-client/mbed-client-classic" from "ssh://git@github.com/ARMmbed/mbed-client-classic.git" at rev #656f13cd4864
hasnain@ubuntu ~/mbed-os/oob-mbed-os-example-client (mbed-client-rc=)
$ cd mbed-os/
hasnain@ubuntu ~/mbed-os/oob-mbed-os-example-client/mbed-os (mbed-os-5.2<)
$ gitk
hasnain@ubuntu ~/mbed-os/oob-mbed-os-example-client/mbed-os (mbed-os-5.2<)
$ git log --oneline | head -1
**a551faa (HEAD, tag: mbed-os-5.2.0-rc2, mbed-os-5.2) Merge pull request #2975 from** ARMmbed/release-candidate
hasnain@ubuntu ~/mbed-os/oob-mbed-os-example-client/mbed-os (mbed-os-5.2<)

mbed-cache messes up the tree.

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

theotherjimmy, 2016-10-20:
@screamerbg ?

@hasnainvirk I think that this is an mbed-cli issue.

bridadan, 2017-01-17:
This is definitely an mbed-cli issue. @hasnainvirk if this problem still persists please file it against https://github.com/ARMmbed/mbed-cli. Thanks!

closed 2017-01-17 by bridadan