ARMmbed/mbed-os #3045: Random Number Generator support

nuket, 2016-10-17:
### Description
- Type: Enhancement
- Priority: Major

**Target**
Nordic nRF5x and any other chip with hardware RNG support

----------------------------------------------------------------
## Enhancement

Are there any plans to design / implement an mbed OS API to access the Random Number Generator (RNG) available on Nordic and other chips? I'm trying to seed the mbed TLS CTR-DRBG implementation, but have no access to a good source of entropy. I've tried about all I can to use the hardware RNG unit, but nothing has worked, maybe someone from Nordic can jump in on this, maybe I'm forgetting to power up or enable the particular hardware unit.

Any attempt to perform write operations on the NRF_RNG register, as in the following code, causes the chip to infinitely reset immediately at start (as I try to use the RNG during the static initialization of a cryptographic Salt class):

```
// Use Nordic Random Number Generator hardware unit.
//
// FIXME: This should be run through mbed TLS CTR-DRBG,
//        there is no guarantee that the Nordic Hardware RNG is 
//        cryptographically strong.
void generate_random_data(uint8_t * output, uint8_t length)
{
    static bool                     initialized = false;
    static mbedtls_ctr_drbg_context context = {0};
    
    if (!initialized)
    {
        // POWER register is no longer present on nrF52, 
        // use START and STOP tasks.
        // NRF_RNG->POWER    = 1;
        
        // Disable interrupt on VALRDY.
        NRF_RNG->INTENCLR = 1;
        
        // mbedtls_ctr_drbg_init(&context);
        // mbedtls_ctr_drbg_seed(&context, nullptr, nullptr, nullptr 0);
        
        initialized = true;
    }

    // NRF_RNG->TASKS_START   = 1;  // Start RNG
    // NRF_RNG->CONFIG        = 1;  // Enable bias correction.
    
    for (uint8_t i = 0; i < length; i++)
    {
        // Trying to write this value on an nRF52 causes immediate reset.
        // NRF_RNG->EVENTS_VALRDY = 0;
        
        // Wait until the value ready event is generated.
        // Roughly 120us, when bias correction is enabled.
        // while ( !NRF_RNG->EVENTS_VALRDY );
        
        nrf_delay_us(250);
        
        output[i] = (uint8_t) NRF_RNG->VALUE;
    }
    
    // NRF_RNG->TASKS_STOP = 1;
}
```

pan-, 2016-10-17:
> Are there any plans to design / implement an mbed OS API to access the Random Number Generator (RNG) available on Nordic and other chips?

Yes!

> Any attempt to perform write operations on the NRF_RNG register, as in the following code, causes the chip to infinitely reset immediately at start (as I try to use the RNG during the static initialization of a cryptographic Salt class):

- Which NRF_RNG cause the hardfault ? 
- Is the softdevice already enabled at this stage ? 
- Did you try to use the functions exposed by the softdevice ?
  * `sd_rand_application_pool_capacity_get`
  * `sd_rand_application_bytes_available_get`
  * `sd_rand_application_vector_get`

nuket, 2016-10-18:
Awesome.
- I've been using the NRF52_DK for most of my prototyping work. The NRF51_DK RNG seemed to work, but had other problems (too little memory to run everything + crypto from mbed TLS).
- The SoftDevice probably is not started at this point, as the code calling into the RNG is getting called from the constructor of one of my classes, which is defined at the head of my main source file. I may try to defer use of the RNG to the point at which the bleInitComplete callback gets called.
- I have not tried the functions exposed by the SoftDevice, but that is definitely a suggestion I will try.

pan-, 2016-10-18:
@nuket If you download nordic SDK V11, you will see a module handling the RNG. If the softdevice is present (which is always the case for now in mbed-OS), then the module relies on the softdevice primitives otherwise it use the registers available. That might give you *inspiration* if you look at this module until we add the support in mbed-os.

nuket, 2016-10-18:
Ahh yes, I was looking at the `nrf_drv_rng.c` file from SDK 12 yesterday. I'll take a closer look at the `sd_rand_*` functions.

nuket, 2016-10-18:
Awesome, the `sd_rand_*` functions work fine.

The Nordic SoftDevice RNG returns 32 bytes at a time. Unsure whether bias correction is enabled. In order to get more than 32 bytes of random data for the CTR-DRBG in SHA-512 mode, when `MBEDTLS_CTR_DRBG_ENTROPY_LEN == 48`, the entropy-gathering user callback needs to pool bytes, by calling `sd_rand_application_vector_get()` multiple times, while waiting between calls for the random number pool to refill.

nuket, 2016-10-19:
Spoke a little too soon. 

The SoftDevice RNG seems to work fine, but attempting to seed the CTR-DRBG leads to a Hard Fault (blinkenlights!), which prints this on the console just before dying:

> RTX error code: 0x00000001, task ID: 0x200073D8

Which seems to indicate [a stack overflow](http://www.keil.com/support/man/docs/rlarm/rlarm_ar_cfgerrfunc.htm)?

```
$ grep -rn OS_ERR_STK_OVF *
mbed-os/rtos/rtx/TARGET_CORTEX_A/rt_System.c:330:    os_error (OS_ERR_STK_OVF);
mbed-os/rtos/rtx/TARGET_CORTEX_A/RTX_Config.h:37:#define OS_ERR_STK_OVF          1
mbed-os/rtos/rtx/TARGET_CORTEX_M/rt_System.c:318:        os_error (OS_ERR_STK_OVF);
mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Config.h:37:#define OS_ERR_STK_OVF          1U
mbed-os/rtos/rtx/TARGET_ARM7/rt_System.c:285:            os_error (OS_ERR_STK_OVF);
mbed-os/rtos/rtx/TARGET_ARM7/RTX_Conf.h:37:#define OS_ERR_STK_OVF          1
```

Based on comments in #2635, the default RTX stack size looks like it's [200 bytes](https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c#L64):

```
//   <o>Default Thread stack size [bytes] <64-4096:8><#/4>
//   <i> Defines default stack size for threads with osThreadDef stacksz = 0
//   <i> Default: 200
#ifndef OS_STKSIZE
 #define OS_STKSIZE     200
#endif
```

Problem is that there's an allocation of 384 bytes in `mbedtls_ctr_drbg_reseed()`, which probably blows everything up:

```
int mbedtls_ctr_drbg_reseed( mbedtls_ctr_drbg_context *ctx,
                     const unsigned char *additional, size_t len )
{
    unsigned char seed[MBEDTLS_CTR_DRBG_MAX_SEED_INPUT];
    ...
}
```

I'm going to try setting a larger stack size for my target and see if that fixes the problem.

For reference, I modified the [BLE_HeartRate example main.cpp](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_HeartRate/source/main.cpp) to look like this, to test my code in near-isolation, but with the SoftDevice running:

```
/* mbed Microcontroller Library
 * Copyright (c) 2006-2015 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include <mbed-events/events.h>
#include <mbed.h>

#include "ble/BLE.h"
#include "ble/Gap.h"
#include "ble/services/HeartRateService.h"

#include "nrf_soc.h"

[... 161 lines cut ...]
    ble.gap().startAdvertising();
}

void scheduleBleEventsProcessing(BLE::OnEventsToProcessCallbackContext* context) {
    BLE &ble = BLE::Instance();
    eventQueue.post(Callback<void()>(&ble, &BLE::processEvents));
}

int main()
{
    printf("main() starting.\r\n");

    eventQueue.post_every(1000, periodicCallback);

    BLE &ble = BLE::Instance();
    ble.onEventsToProcess(scheduleBleEventsProcessing);
    ble.init(bleInitComplete);

    while (true) {
        eventQueue.dispatch();
    }

    return 0;
}
```

nuket, 2016-10-19:
Hmm. Well, that didn't work. Setting `OS_IDLESTKSIZE=2000`, `OS_STKSIZE=2000`, `OS_MAINSTKSIZE=2000` in `mbed_app.json` still sees the above program jumping to a HardFault pretty much right after `main() starting.` is printed.

nuket, 2016-11-24:
Still unsure why this HardFault is happening.

I instrumented mbedtls_ctr_drbg_reseed to look like:

```
int mbedtls_ctr_drbg_reseed( mbedtls_ctr_drbg_context *ctx,
                     const unsigned char *additional, size_t len )
{
    unsigned char seed[MBEDTLS_CTR_DRBG_MAX_SEED_INPUT];
    size_t seedlen = 0;

    printf("mbedtls_ctr_drbg_reseed 1\r\n");
    
    if( ctx->entropy_len + len > MBEDTLS_CTR_DRBG_MAX_SEED_INPUT )
        return( MBEDTLS_ERR_CTR_DRBG_INPUT_TOO_BIG );

    memset( seed, 0, MBEDTLS_CTR_DRBG_MAX_SEED_INPUT );

    /*
     * Gather entropy_len bytes of entropy to seed state
     */
    if( 0 != ctx->f_entropy( ctx->p_entropy, seed,
                             ctx->entropy_len ) )
    {
        return( MBEDTLS_ERR_CTR_DRBG_ENTROPY_SOURCE_FAILED );
    }

    seedlen += ctx->entropy_len;

    printf("mbedtls_ctr_drbg_reseed 2\r\n");
    
    /*
     * Add additional data
     */
    if( additional && len )
    {
        memcpy( seed + seedlen, additional, len );
        seedlen += len;
    }

    /*
     * Reduce to 384 bits
     */
    block_cipher_df( seed, seed, seedlen );

    /*
     * Update state
     */
    ctr_drbg_update_internal( ctx, seed );
    ctx->reseed_counter = 1;

    printf("mbedtls_ctr_drbg_reseed 3\r\n");
    
    return( 0 );
}
```

But it dies right after printing the third line. Is anyone from Nordic following this issue?

nvlsianpu, 2016-11-29:
Hi. Did you have this https://github.com/ARMmbed/mbed-os/pull/2969 included in mbed-os you using for development?

nvlsianpu, 2016-11-29:
BTW It is said that nRF52's RNG generates true non-deterministic random numbers based on internal
thermal noise (by nRF52 data sheet). So there is the warrant of strong capabilities.

seppestas, 2017-08-18:
Is there a reason why the [RNG driver](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v11.0.0/group__nrf__drv__rng.html) is not available in Nordic SDK v11 included in mbed-os? It seems to be there for the nRF51 and v13 of the SDK (only used for the nRF52840).

dlfryar, 2018-01-12:
Mbed OS SDK updates happening [here](https://github.com/ARMmbed/mbed-os/pull/5845)

marcuschangarm, 2018-03-08:
New TRNG driver has been merged to master: https://github.com/ARMmbed/mbed-os/pull/6116

closed 2018-05-18 by 0xc0170