ARMmbed/mbed-os #3049: [Exporters] Uvision5

ohagendorf, 2016-10-17:
### Description
- Type: Bug
- Related issue: PR #2708
- Priority: Blocker

When exporting a project, e.g. with
```
mbed export -m NUCLEO_F207ZG -i uvision5
```
and opening it in uVision5 the project can be compiled and flashed without problems. Nevertheless the project configuration is incomplete, as described under **Steps to reproduce**.

But when you want to change a debuger option, e.g switching on Reset and Run, the project can't be used any longer to flash the mcu. When you try it, an error message appeares which doesn't help to solve the problem. The reason is an empty flash algorithm config. Adding an algorithm is not possible. The offered list doesn't contain the correct entry but several other. Please see the screendump at the end.

The exact behaviour depends on mcu and uvison5 version:
- NUCLEO_F207 and uvision 5.20: Project is completly bricked!! No possibility found to repair it.
- NUCLEO_F767 and uvision 5.20: Project can be repaired, please see below.
- both targets and uvision5 5.21a: Project can be repaired, please see below.

Repairing a project does mean:
- open Options for Targets ...
- change the device to another one
- change the device back to original one
- goto Debug config
- choose ST-Link Debugger
- close with OK
- save all (**without saving the next steps will not work!**)
- open again Options for Targets ...
- goto Debug config
- open Settings dialog
- goto Flash Download
- add the correct flash algorithm from a list of several variants (only one is correct!)
- after all these steps with NUCLEO_F207 and uvision 5.20 it is still not possible to add a flash algorithm - the list doesn't contain the correct line

These steps are not really obvious so changing a debugger setting means bricking the project file!

A similar behavior was already repaired before the above PR. Based on the discussion there maybe it is a problem of STM targets in uVision5. But we didn't examined all targets. Creating the .uvoptx solved the problem at that time. Maybe the current format of this files is the problem.

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_xxx

**Toolchain:**
uvision5

**Toolchain version:**
5.20 and 5.21a

**mbed-cli version:** 
0.9.9

**meed-os sha:**
mbed-os 5.2 rc3
873686c Merge pull request #3033 from ARMmbed/release-candidate
and current master
7bcc63b Merge pull request #2977 from andreaslarssonublox/ublox_drivers

**Expected behavior**
Programming Algorithm should contain one line with the flash algorithm

**Actual behavior**
Programming Algorithm is empty

**Steps to reproduce**
- open Options for Targets ...
- goto Debug config
- open Settings dialog
- goto Flash Download

It looks like this:
[image]

possible algorithms before repairing as described above:
[image]

It should look like this:
[image]

and now Add offers these algorithms:
[image]

ohagendorf, 2016-10-17:
@sarahmarshy, please have a look at the above problem description.

0xc0170, 2016-10-18:
Thank you very much for the detailed description, I have noticed some of the problem with flash algo. We will have a look at this to resolve.

ohagendorf, 2016-10-20:
@theotherjimmy, it is not only a problem of Nucleo boards but also of Discovery boards. Maybe more, I tested exporting and opening in uVision with some other tagrgets, I've never seen an entry in the Programming Algorithm list.

theotherjimmy, 2016-10-20:
I don't think we populate it.

theotherjimmy, 2016-10-20:
So I updated the title to better match the problem.

0xc0170, 2016-10-21:
@theotherjimmy @sarahmarshy How to get those flash algos in the debugger settings?

sg-, 2017-01-19:
Has this been resolved?

closed 2017-10-27 by ChiefBureaucraticOfficer