ARMmbed/mbed-os #3057: Exporting to uvision fails

hasnainvirk, 2016-10-18:
Setup:

    mbed-os-5.2.0-rc3
    mbed-cli 0.9.9 Command used:

mbed export -m NUCLEO_F401RE -i uvision

Error message :

Traceback (most recent call last):
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project.py", line 234, in
main()
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project.py", line 230, in main
zip_proj=zip_proj, build_profile=profile)
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project.py", line 93, in export
build_profile=build_profile, silent=silent)
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project_api.py", line 229, in export_project
macros=macros)
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project_api.py", line 90, in generate_project_files
exporter.generate()
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/export/uvision/init.py", line 192, in generate
'device': DeviceUvision(self.target),
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/export/uvision/init.py", line 23, in init
DeviceCMSIS.init(self, target)
File "/home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/export/cmsis/init.py", line 53, in init
raise TargetNotSupportedException("Target not in CMSIS packs")
tools.export.exporters.TargetNotSupportedException: Target not in CMSIS packs
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/hasnain/mbed-os/oob-mbed-os-example-client/mbed-os/tools/project.py -i uvision -m NUCLEO_F401RE --source ." in "/home/hasnain/mbed-os/oob-mbed-os-example-client"

This is not a Duplicate of https://github.com/ARMmbed/oob-mbed-os-example-client/issues/17. Opened a ticket there for better visibility and quick resolution.

@sg- , @MarceloSalazar , @SeppoTakalo , @0xc0170

ohagendorf, 2016-10-18:
Please check C:\Users\YOURUSERNAME\\.local\share\arm-pack-manager if there is an empty index.json.

When yes, it seems to be the problem I described in #3049 

Maybe you can find a first solution in my description (using an older version of pycurl)

ohagendorf, 2016-10-18:
@hasnainvirk , sorry it is #3051

hasnainvirk, 2016-10-18:
@ohagendorf I am using Linux. I don't think Windows 10 solution applies to me.

ohagendorf, 2016-10-18:
But you can check if the CMSIS packs are available. Unfortunatly I can't say where the files are stored under Linux.
As @bridadan wrote, you have to install libcurl manually under Linux.

theotherjimmy, 2016-10-20:
@hasnainvirk they were stored in $HOME/.local/share/arm-pack-manager. 

OTOH, #3067 was merged, so this should no longer be an issue.

closed 2016-12-20 by theotherjimmy