ARMmbed/mbed-os #3058: Race condition in UDP sendto or recvfrom?

EduardPon, 2016-10-18:
If UDP is blocking mode (_timeout > 0), we presume there is a race condition in the functions UDPSocket::sendto and UDPSocket::recvfrom when a multiple number threads share the socket.

When a thread enters for instance the sendto function, the mutex _lock.lock() is set. If a second thread  calls the sendto also, it will be blocked in the _lock.lock() call. Next the first caller will execute: 
```
_lock.unlock();
////// 2ND CALLER COULD BE UNBLOCKED ///////
count = _write_sem.wait(_timeout);
_lock.lock(); 
```
It can unlock the 2nd caller, but the _read_in_progress is still not cleared. As result the 2nd caller will run in a assert:
`MBED_ASSERT(!_read_in_progress); `

bridadan, 2016-10-18:
Any ideas @geky ?

geky, 2016-10-18:
I believe those assertions were initially put in place specifically to prevent the use of `recv` from multiple threads.

However, looking into this more, I'm not entirely sure why those assertions are necessary. From the [commit](https://github.com/ARMmbed/mbed-os/commit/04e4d9f450dfcc2b87280c7bb1436177056c6a93) it looks like the goal is to avoid interleaved data from a recv call. However this would be the expected result of calling `recv` from multiple threads. @c1728p9 is there another reason for not removing these assertions?

c1728p9, 2016-10-18:
The assertions were added when the mutexes protecting multiple reads or multiple writes at the same time were removed. For UDP, send guarantees that either the whole packet is sent or nothing is sent, so the assert there should be removed here. The same argument could be made for UDP receive. For TCP send an receive calls should never be made at the same time since the data can interleave. Using TCP in this manner is an application logic error so the asserts here should be kept.

geky, 2016-10-18:
If the user can garuntee that their payload is less than the MTU accross the entire connection, interleaving can be avoided for TCP. A good example would be an application that sends small 4-byte updates from multiple threads, but uses TCP to handle packet-loss and keep-alive.

theotherjimmy, 2016-12-20:
@geky @c1728p9 #3065 was merged can we close this now?

c1728p9, 2016-12-20:
Fine by me

geky, 2016-12-20:
Same, I don't have a preference

closed 2016-12-21 by 0xc0170