ARMmbed/mbed-os #3071: TLS link time errors when no entropy source

c1728p9, 2016-10-18:
Prior to PR #2843 security features were turned on even when null entropy was not. This allowed example programs with a link time dependency on TLS but did not actually use TLS to compile and work. One example of this is https://github.com/ARMmbed/mbed-os-example-mesh-minimal.

After PR #2843 examples like this started having linker issues on IAR and ARM since these compilers are less permissive about missing functions, even if they are never referenced. This causes confusing errors to users since it is not obvious that the culprit is a missing entropy source.

TLS should either provide stubs for these functions that have been compiled out, or make it an error to include TLS headers with prototypes that have their definitions turned off.

CC: @andresag01

andresag01, 2016-10-19:
I raised an issue in mbedTLS to address this problem https://github.com/ARMmbed/mbedtls/issues/660.

sg-, 2017-01-19:
Has this been resolved?

sbutcher-arm, 2017-01-20:
This bug was trying to generalise a solution for a specific bug in the mbed OS example for mesh minimal - "[Cannot build for ARMCC on boards without TRNG #24](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/issues/24)".

The specific bug was fixed long ago in "[Fix mbedtls user config and null entropy #2926]"(https://github.com/ARMmbed/mbed-os/pull/2926), which changed the way the configurations were done.

However, the general problem which @c1728p9 described as ‘TLS should either provide stubs for these functions that have been compiled out, or make it an error to include TLS headers with prototypes that have their definitions turned off’ is not done, and is still pending work as the mbed TLS issue "[Function declarations not conditionally compiled #660](https://github.com/ARMmbed/mbedtls/issues/660)".

Because there is no specific or immediate problem for mbed OS, and because the issue is raised as a separate issue in the mbed TLS repo as bug #660, I see no need to keep this one open if you want to close it.

bridadan, 2017-01-20:
Thanks for the update @sbutcher-arm, I'll go ahead and close this then.

closed 2017-01-20 by bridadan