ARMmbed/mbed-os #3081: How to track down task ID's of RX errors?

EduardPon, 2016-10-20:
Is there a way to log the ID's and reference where this is done of task creation (threads/timers)?

Currently it is hard to track down which task ID runs into a RTX error.

0xc0170, 2016-10-20:
cc @c1728p9 @geky

geky, 2016-10-21:
You can use [`Thread::gettid`](https://github.com/ARMmbed/mbed-os/blob/master/rtos/Thread.h#L315-L318) function to access the thread's id when you spawn a thread. I believe this is the same id printed during errors.

I do know that the thread id is just a casted pointer to rtx's TCB structure ([here](https://github.com/ARMmbed/mbed-os/blob/3a16ca9855528fdb801b11f275ffdca7b5ed9f26/rtos/rtx/TARGET_CORTEX_M/rt_TypeDef.h#L57)). If you have a debugger connected you can get a hold of some pretty useful information.

For example with gdb:
``` gdb
set $tcb = (P_TCB)0x1b2b34
# The entry point of the thread
print/a $tcb->ptask    
# The current state of the thread
print (Thread::State)$tcb->state
# Recently called function
print/a ((int*)$tcb->tsk_stack)[13]
# Current stack pointer
print/a $tcb->tsk_stack
```

bridadan, 2017-01-17:
I'm guessing that @geky's answer solved your issue so I'll go ahead and close this. Please reopen it if you'd like to continue the discussion!

closed 2017-01-17 by bridadan