ARMmbed/mbed-os #3082: Queue overflow by system tick event

EduardPon, 2016-10-20:
The sysTimerTick tries to put an event into kernel timer queue (isrMessagePut function   cortex M  rt_CMSIS.c), but overflows by performing the following sequence on our K64F boards:

- Perform IP UDP communication (TX CoAP messages addressed local host and multi cast).
- Unplug the Ethernet cable and wait at least 5 seconds. UDP communication is in the meanwhile not stopped.
- Plug in the Ethernet cable. Link up again signaled.
- The overflow occurs.... 

What is the issue when the kernel timer queue overflows?

adbridge, 2016-11-03:
cc @geky @c1728p9

EduardPon, 2016-11-09:
Seems not to occur anymore after transition to mbed5.2.1.
I will do some more testing the next coming weeks before I close this issue.

bridadan, 2017-01-17:
@EduardPon Did you ever have a chance to investigate this some more?

EduardPon, 2017-01-18:
No, I didn't. Because it is not biting anymore in mbed5.2.1.

bridadan, 2017-01-18:
Ok, in that case I'll go ahead and close this. Please reopen this though if you run into it again!

closed 2017-01-18 by bridadan