ARMmbed/mbed-os #3091: How to check version of mbed-os at compile time?

infinnovation, 2016-10-20:
### Description
- Type: Question

## Question

Is there a way to check which version of mbed-os is present e.g. 5.1 versus 5.2?  It would be nice if there were some macro such as MBED_VERSION to check against, but I don't see one.

I was trying to get some code to compile cleanly against both 5.1 and (release candidate) 5.2.  In 5.2, some methods are deprecated, resulting in warnings, but the alternatives are not available in 5.1.  For example Callback(func, arg) c.f. Callback(arg, func).

Sure, I can find some other shibboleth such as MBED_CONF_STDIO_BAUD_RATE (renamed to MBED_CONF_PLATFORM_STDIO_RATE in 5.2) but that doesn't smell right.

0xc0170, 2016-10-20:
We provide mbed library macro for mbed 2. I think it would be nice to have a version in the code (x.y.z)

cc @sg-

0xc0170, 2016-11-09:
Fixed via https://github.com/ARMmbed/mbed-os/pull/3207

closed 2016-11-09 by 0xc0170