ARMmbed/mbed-os #3100: [UBLOX_EVK_ODIN_W2] Export to make_iar fails to compile - 5.2-rc4

bulislaw, 2016-10-20:
### Description
- Type: Bug
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
IAR

**meed-os sha:**
1d87508

It works with ```mbed compile -t iar``` but fails with export to make_iar:
```
Errors: none
Warnings: none
"link: oob-mbed-os-example-wifi.elf"

   IAR ELF Linker V7.70.2.11706/W32 for ARM
   Copyright 2007-2016 IAR Systems AB.
Fatal error[Ms007]: could not open file "libublox-odin-w2-driver.a"
            Directories searched:
              C:\Users\root\bartek\oob-mbed-os-example-wifi\.build
              C:\Users\root\bartek\oob-mbed-os-example-wifi\mbed-os\targets\TAR
                      GET_STM\TARGET_STM32F4\TARGET_UBLOX_EVK_ODIN_W2\sdk\TOOLC
                      HAIN_IAR
Fatal error detected, aborting.
make[1]: *** [oob-mbed-os-example-wifi.elf] Error 3
make: *** [all] Error 2
```

The .a file exists but the actual name is ublox-odin-w2-driver.a

0xc0170, 2016-10-24:
cc @theotherjimmy @sarahmarshy

theotherjimmy, 2016-12-20:
This has been fixed by @bridadan with PR #3243 and is now in 5.3.0. Closing.

closed 2016-12-20 by theotherjimmy