ARMmbed/mbed-os #3103: Makefiles should assume your compiler is in your path

bridadan, 2016-10-21:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
Currently Makefiles hard code the paths to the compilers based on the paths set by `mbed_settings.py` and `settings.py`. This is not always the correct path, especially if you have the compiler in your path already.

**Suggested enhancement**
Since `make` is already required to be in your path, for Makefiles we could require that the compilers are in your path as well. The Makefile should probably print a small message at the beginning of the compile process saying that the compiler is expected be in the path.

**Pros**
- Simple!
- Should work ok in the online system as well

**Cons**
- One more setup step for users that don't put compilers in their system path.

Any thoughts on this @theotherjimmy, @sarahmarshy, @0xc0170?

sarahmarshy, 2016-10-22:
Yes, getting rid of these lines https://github.com/ARMmbed/mbed-os/blob/master/tools/settings.py#L40 would be great. We already assume things about where users are putting their installations, and I think it would be better to assume they are in the path rather than these arbitrary locations. The con can be mitigated because setup already requires an additional step of editing mbed_settings.py.

bulislaw, 2016-10-24:
Yes please, that would be great!

bridadan, 2016-12-13:
This has been in master for sometime now, closing!

closed 2016-12-13 by bridadan