ARMmbed/mbed-os #3105: Makefile for nrf5x - combined requirement not documented

0xc0170, 2016-10-21:
### Description
- Type: Enhancement 
- Priority: Major

## Enhancement

**Reason to enhance or problem with existing solution**

The mbed-cli produces hex file that contains softdevice, thus a user can copy this binary output and it works. However, makefile produces also -combined.hex that is most likely not documented. We should be consistent

**Suggested enhancement**

Either use suffix combined for all tools, or remove that suffix. So a user can switch between tools and use the same steps.

@theotherjimmy @pan- @ashok-rao @MarceloSalazar

theotherjimmy, 2016-12-20:
We are part of the way there: when a makefile produces the final product, it informs the user now, like the command line does.

bridadan, 2016-12-20:
Do we know which naming convention we want to use? I'd argue we should go with whatever one we have the least control over. So if IAR or uVision already use one of the naming conventions and we can't change that naming, we should probably just go with that?

theotherjimmy, 2016-12-20:
I'm pretty sure that IAR and uVision don't do the merging...

bridadan, 2016-12-20:
Any input on this @pan-? I'm honestly sure what IAR and uVision do either :smile:

pan-, 2016-12-21:
@theotherjimmy You are correct, IAR and uVision don't do the merging but because they load the application at the correct memory offset and don't erase the whole flash, it is possible to work with those tools.

It would be nice to have the exported project loading the softdevice during download **and** generating a full image in post build stage.

closed 2017-10-27 by ChiefBureaucraticOfficer