ARMmbed/mbed-os #3106: Empty destructors code of HW drivers

nvlsianpu, 2016-10-21:
### Description
- Type: Bug

Most of driver has empty destructor code. The consequence is that there is no possible to de-initialize corresponding hardware in uC. It might caused malfunction when uC common hw part (e.g. GPIO) were used for few  peripherial or so. It also might cause incresed power consumption because there is no entry to disable Hardware.

For example on nRF5 target we can use a pin as AnalogIn an after that as DigitalInOut. Once a pin was configured as analog pin, such a pin was assigned to the  ADC and the Digital part was disabled. Because of lack of destructor there is no entry to de-initialize analog part of a pin HW or disabling ADC at all. This cause that this pin will be unable to work well if we try to reuse it for DigitalInOut.

example of use case:
https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/AnalogIn/AnalogIn.cpp

BlackstoneEngineering, 2016-10-21:
Would you be interested in writing a test that could test for this and adding it to the [CI Test Shield](http://github.com/ARMmbed/ci-test-shield) Project?

@sg- @screamerbg

nvlsianpu, 2016-10-21:
cc @pan-, @anangl

bcostm, 2016-10-22:
Also here #2017

pan-, 2016-10-26:
Thanks for raising this issue, lack of destructors is present in many driver classes while on others the destructor declared doesn't release any resources. Adding proper release process also means that copy construction and copy assignment should be properly handled, either by forbidding such operations or by reference counting. 

This is the list of the classes impacted by these issues:

|                                                                                               | Destructor | Copy/Assignment strategy | Note                                                                                                                                                     |
|-----------------------------------------------------------------------------------------------|:----------:|:------------------------:|----------------------------------------------------------------------------------------------------------------------------------------------------------|
| [`AnalogIn`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/AnalogIn.h)               |      ✖     |             ✖            |                                                                                                                                                          |
| [`AnalogOut`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/AnalogOut.h)             |      ✖     |             ✖            |                                                                                                                                                          |
| [`BusInOut`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/BusInOut.h)               |      ✔     |             ✖            | Copy assignment operator declared twice (one public and one private), the public one doesn't make a copy at all and is actually a shorthand for `write`. |
| [`CAN`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/CAN.h)                         |      ✔     |             ✖            |                                                                                                                                                          |
| [`DigitalIn`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/DigitalIn.h)             |      ✖     |             ✖            |                                                                                                                                                          |
| [`DigitalInOut`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/DigitalInOut.h)       |      ✖     |             ✖            |                                                                                                                                                          |
| [`DigitalOut`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/DigitalOut.h)           |      ✖     |             ✖            |                                                                                                                                                          |
| [`Ethernet`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/Ethernet.h)               |      ✔     |             ✖            |                                                                                                                                                          |
| [`I2C`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/AnalogIn.h)                    |      ✖     |             ✖            |                                                                                                                                                          |
| [`I2CSlave`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2CSlave.h)               |      ✖     |             ✖            |                                                                                                                                                          |
| [`InterruptIn`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/InterruptIn.h)         |      ✖     |             ✖            |                                                                                                                                                          |
| [`LowPowerTicker`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/LowPowerTicker.h)   |      ✔     |             ✖            |                                                                                                                                                          |
| [`LowPowerTimeout`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/LowPowerTimeout.h) |      ✔     |             ✖            |                                                                                                                                                          |
| [`LowPowerTimer`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/LowPowerTimer.h)     |      ✖     |             ✖            |                                                                                                                                                          |
| [`PortIn`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/PortIn.h)                   |      ✖     |             ✖            |                                                                                                                                                          |
[... cut ...]
| [`Timer.h`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/Timer.h)                   |      ✖     |             ✖            |                                                                                                                                                          |
| [`TimerEvent.h`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/TimerEvent.h)         | ✖          |             ✖            | Destructor just remove the registered function; it should shutdown the timer if the timer is not used anywhere else.                                     |

seppestas, 2017-03-17:
As mentioned in #3945 this issue also affects the use of the [improved mbed sleep API](https://developer.mbed.org/teams/SiliconLabs/wiki/Using-the-improved-mbed-sleep-API).

The `LPC176X` target doesn't do anything in it's `pwmout_free` implementation, but e.g Silicon Lab's `EFM32` targets requires e.g the `pwmout_free` function to be called to unlock low power sleep modes when the `sleep` function is used. This means the low power sleep will not work when a `PwmOut` object has been initialized, even when it has been deconstructed.

Personally I'm more a fan of some kind of platform independant "disable" function to disable the PWM clock and enable the low power sleep modes again. The same goes for other drivers / peripherals. Currently [constantly destructing PWM objects before sleeping and reconstructing them on wakeup seems to be "the way to do it"](https://developer.mbed.org/questions/998/Deactivate-PWM/). Not the biggest fan of this approach though, and I have yet to figure out if this causes memory fragmentation issues.

seppestas, 2017-04-04:
Would PR's solving parts of this issue get accepted? Maybe it's a good idea to start a feature branch for this?

Low power operation is a must for my use-case (and I imagine many others) and being able to disable hardware is important to accomplish this. This issue is currently the biggest problem for most our projects using mbed, requiring custom patches to mbed and thus greatly increasing complexity (e.g relying on mbed forks, making it harder to do things like update the mbed version).

0xc0170, 2017-04-04:
> Would PR's solving parts of this issue get accepted? Maybe it's a good idea to start a feature branch for this?

Yes, they could. Please provide tests with any patches. You can also create a new issue or just PR with a description of the problem and a proposal to fix (=solving those parts one by one).

We can create a feature branch if needed.

seppestas, 2017-04-04:
Why did #2116 (the oldest still open PR of mbed-os by the way) not get merged?

0xc0170, 2017-04-04:
> Why did #2116 (the oldest still open PR of mbed-os by the way) not get merged?

As stated there (at least partially). We need to work on the documentation (specifications, what _free should do ?) and tests - to comply with the specifications, and fix all targets. As you noticed some targets have free empty or might not even be implemented.

seppestas, 2017-08-03:
So, any updates on this?

The ADC of the nRF53832 SoC (TARGET_NRF5) seems to consume quite a lot of power when it's not being used because of this. It's pretty much impossible to fix this without breaking mbed's HAL.

0xc0170, 2017-08-03:
Not at the moment, we are looking at HAL, so will get better specification (also regarding init/free). 

You can for now as you noted, add free function call and make it work within nrf5 family.

seppestas, 2018-03-23:
I'm bumping this issue again. Not being able to free pins, disable clocks and re-enable sleep modes is a real pain when trying to develop low-power and/or dynamic applications.

We are slowly gathering project/target specific patches to circumvent this issue that we are not able to share since there is still no specification on disabling / freeing drivers. This makes it harder for us to update our mbed project and has a negative effect on the quality of mbed in general.

0xc0170, 2021-12-13:
I am reopining this as it's tracking issue that is still valid.