ARMmbed/mbed-os #3107: Misleading readme for mbed-mesh-api

tommikas, 2016-10-21:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------

## Bug

Channel range is misleading. At least channels beyond 26 don't seem to work.

| 6lowpan-nd-channel        | number [0-**27**] | RF channel to use when `channel_mask` is not defined |

**Expected behavior**

Using values in the range given in the [readme](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/README.md) provides a working configuration.

**Actual behavior**

They don't.

**Steps to reproduce**

Try to use e.g. channel 27 for the mesh interface.

0xc0170, 2016-10-24:
@tommikas Can you send a fix please for this readme?

0xc0170, 2016-10-24:
cc @SeppoTakalo

tommikas, 2016-10-24:
Unfortunately I don't really know enough about this to know what it should be, or whether there are other inaccuracies. Just happened to run into this particular problem.

SeppoTakalo, 2016-10-24:
If using channel page 0:
For 2.4 Ghz I believe channel numbers go from 11 to 26.
For 868 Mhz channel 1 is only allowed.
For 915 Mhz channels 2 to 10 are valid.

So maybe we need to remove the `[0..27]` instruction because it all depends on your frequency.

closed 2017-10-27 by ChiefBureaucraticOfficer