ARMmbed/mbed-os #3115: Issue of when using "mbed.lib" and "mbed-rtos.lib" at the same time.

d-kato, 2016-10-24:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
GR-PEACH, NUCLEO-F429ZI

**Toolchain:**
ARM (Online compiler)

**Expected behavior**
LED blink.

**Actual behavior**
The program to crash with the following debug message:
_mbed assertation failed: interrupts_disabled, file: C:\Jenkins\workspace\mbed-2-build-library\hal\common\mbed_critical.c, line 54_

**Steps to reproduce**
Run the following program.
https://developer.mbed.org/users/dkato/code/test_prj_2/file/7988ca23bda3/main.cpp

When "mbed-rtos.lib" is removed, it work. Or, when using the "mbed-dev.lib" instead of "mbed.lib", it work.

**Consideration**
There is a possibility that a problem occurs to the occasion using the "PlatformMutex" class and "mbed-rtos.lib". The following static memory is in the AnalogIn class.

_static SingletonPtr<PlatformMutex> _mutex;_

This memory is static, so the size is decided at the time of "mbed.lib" build. The size of the "PlatformMutex class" will differ on whether the "MBED_CONF_RTOS_PRESE" has been defined.
When "mbed.lib" created, "MBED_CONF_RTOS_PRESE" is undefined. When you build the sample, the value of the "MBED_CONF_RTOS_PRESE" is determined by whether or not to link the "mbed-rtos.lib".
Therefore, I think that does not work properly in the sample that contains the "mbed-rtos.lib".

toyowata, 2016-10-24:
I just quick tested it and I also reproduced this runtime assertion failed by following platforms:

* LPC1768
* FRDM-K64F
* nRF52-DK
* LPC4088-QSB

I guess all mbed2 + mbed-rtos lib combinations have same issue.

cc @sg- @0xc0170

0xc0170, 2016-10-24:
cc @c1728p9

c1728p9, 2016-10-24:
Hi @d-kato thanks for the example program. It made this problem easy to reproduce. I can confirm that this is a problem due to the memory size of PlatformMutex changing depending on if MBED_CONF_RTOS_PRESENT is defined. Since the mbed library is built with MBED_CONF_RTOS_PRESENT undefined, this value should also be undefined when building a user application.

The fix I propose is to delete mbed-rtos/mbed_lib.json, since MBED_CONF_RTOS_PRESENT should never be defined for 2.0 builds. @toyowata, @d-kato can you try deleting this file and checking if this fixes the problem? If it does I'll update mbed-rtos.

toyowata, 2016-10-25:
Hi @c1728p9 thank you very much for your quick investigation.

>can you try deleting this file and checking if this fixes the problem? If it does I'll update mbed-rtos.

I couldn't see assertions on multiple target boards by removing mbed_.lib.json in mbed-rtos library.

d-kato, 2016-10-25:
Hi @c1728p9. Does exclusive access control of multiple tasks work by this change?

c1728p9, 2016-10-25:
Mutexes work as expected, but PlatformMutex provides no protection. Everything needs to be protected by the application with a Mutex.

d-kato, 2016-10-25:
I understand. Thank you.

0xc0170, 2016-10-25:
@c1728p9 Let us know once the rtos is fixed as proposed above, and we shall close this issue as resolved

c1728p9, 2016-11-09:
The RTOS has been updated so this issue should now be fixed.

d-kato, 2016-11-10:
I confirmed that it has been fixed.

closed 2016-11-10 by d-kato