ARMmbed/mbed-os #3117: Exporting using project.py in the offline version is not working

kgoveas, 2016-10-24:
### Description
- Type: Bug 
- Related issue: `unknown`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
LPC1768 - All Supported MCUs

**Toolchain:**
GCC_ARM

**Toolchain version:**
eclipse_gcc_arm, make_gcc_arm, gcc_arm

**mbed-os branch**
(`https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.2`)

**DAPLink version:**

**Expected behavior**
Expected to provide the source library which can be compiled using eclipse-makefile based system like former releases.
**Actual behavior**
```
D:\mbed-os-mbed-os-5.2\tools>python project.py -i eclipse_gcc_arm -p 0
Scan: basic
Scan: env
Scan: hal
Traceback (most recent call last):
  File "project.py", line 244, in <module>
    main()
  File "project.py", line 240, in main
    zip_proj=zip_proj, build_profile=profile)
  File "project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "D:\mbed-os-mbed-os-5.2\tools\project_api.py", line 229, in export_project
    macros=macros)
  File "D:\mbed-os-mbed-os-5.2\tools\project_api.py", line 90, in generate_project_files
    exporter.generate()
  File "D:\mbed-os-mbed-os-5.2\tools\export\cdt\__init__.py", line 14, in generate
    super(Eclipse, self).generate()
  File "D:\mbed-os-mbed-os-5.2\tools\export\makefile\__init__.py", line 82, in generate
    ctx[key] = ctx['vpath'][0] + "/" + ctx[key]
TypeError: cannot concatenate 'str' and 'NoneType' objects
```
**Steps to reproduce**
In the tools directory, using the command as mentioned below doesn't work.
``` python project.py -i eclipse_gcc_arm -p 0 ```

0xc0170, 2016-10-24:
I can reproduce this locally, from master.

cc @theotherjimmy @sarahmarshy

kgoveas, 2016-10-24:
Hi @0xc0170,

Is there a current plan to support the offline toolchain? 

Also some documentation regarding offline exporting would be helpful around build.py, and project.py. 

Thank you.

0xc0170, 2016-10-24:
> Also some documentation regarding offline exporting would be helpful around build.py, and project.py.

Please request the docs in the new issue, for visibility.

>Is there a current plan to support the offline toolchain?

If you are asking about this issue, will be fixed

kgoveas, 2016-10-24:
Ok. Thank you.

0xc0170, 2016-10-28:
bump @theotherjimmy

theotherjimmy, 2016-11-02:
You don't have a linker script. I should make that error a bit more reasonable.

theotherjimmy, 2016-11-02:
More digging reveals something interesting: The restructure PR split the hal into the targets directory and the hal directory. Tests still assume that they only need the hal directory when they need both. This manifests itself when the exporter does not find a liker script, at they are contained within the targets directory.

theotherjimmy, 2016-11-02:
PR up!

theotherjimmy, 2016-12-20:
@kgoveas @0xc0170 The mentioned issue is fixed, but I'm going to add a check real quick to improve that error message.

closed 2016-12-20 by theotherjimmy