ARMmbed/mbed-os #3119: nRF51822 flash overflow not detected

Timmmm, 2016-10-24:
I'm using mbed-os 5 with the nRF51-DK target. This makes it use `mbed-os\hal\targets\cmsis\TARGET_NORDIC\TARGET_NRF5\TARGET_MCU_NRF51822_UNIFIED\TOOLCHAIN_GCC_ARM\TARGET_MCU_NORDIC_32K\NRF51822.ld` as the linker script.

I'm very close to the flash size limit, but before I actually reach the `region `FLASH' overflowed by 4704 bytes` error there comes a point where it still compiles, yet the hex file goes past the 256 kB limit. This means you get this error:

    ERROR: The file specified is not a valid hex file, has data outside valid areas
    ERROR: or does not have data in valid areas.

I verified by opening the .hex file in J-Flash that my hex file goes 44 bytes past 0x40000. Here is the first part of the linker script and the maths seems correct so to be honest I'm not sure why this happens. But it does.

```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x0001B000, LENGTH = 0x25000
  RAM (rwx) :  ORIGIN = 0x20002ef8, LENGTH = 0x5108
}
```

0xc0170, 2016-10-25:
cc @pan-

0xc0170, 2016-10-25:
cc @pan- @nvlsianpu @anangl

sg-, 2017-01-19:
What version toolchain are you using?

theotherjimmy, 2017-04-26:
@Timmmm Could you respond to @sg- or close the issue if this is no longer a problem?

Timmmm, 2017-04-27:
I can't remember to be honest, and have also switched jobs and don't do much mBed stuff any more unfortunately so I will close this. :-)

closed 2017-04-27 by Timmmm