ARMmbed/mbed-os #3123: "target_overrides" unhandy for "macros" with values

betzw, 2016-10-25:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
When trying to override the value of a macro defined in `targets.json` for an application in file `mbed_app.json` I didn't find a better way then first removing the first macro **including** its value and then add the macro with its new value (all my other trials either didn't work or produced ugly compiler warnings), e.g. like this:
```
{
    "target_overrides": {
        "*": {
           "target.macros_remove": ["TRANSACTION_QUEUE_SIZE_SPI=2"],
           "target.macros_add": ["TRANSACTION_QUEUE_SIZE_SPI=0"]
       }
    }
}
```

**Suggested enhancement**
The unfortunate things here are that we need two lines to express what we want to do and above all that we need to know also the original value of the macro. Unfortunately, this value could even be different from platform to platform, which would require to specify the overrides for each platform we are targeting in our application!
Therefore, I would propose some additional syntax like this:
```
{
    "target_overrides": {
        "*": {
           "target.macros_value": ["TRANSACTION_QUEUE_SIZE_SPI=0"]       
        }
    }
}
```

**Pros**
- No need to know & specify the original value, which might be different from platform to platform and therefore we would need to differentiate even more 
- Simpler syntax

0xc0170, 2016-10-25:
@bogdanm @theotherjimmy

bogdanm, 2016-10-25:
The purpose of `macros_add` was to allow for **cumulative** values: you *add* this macro to the list of macros you inherit from your base target(s) (same goes for `remove`). When you use `target.macro_values`, is that still cumulative, or does it just set the macro values (without inheriting the other macros from the base target(s))? It feels a bit strange. We could introduce `macros_replace` to combine `macro_add` and `macros_remove` in a single operation, would that help?

betzw, 2016-10-25:
The operation would not be cumulative but is intended to **replace** the value assigned to an inherited macro by another value without having to specify the original value of the inherited macro.

bogdanm, 2016-10-25:
I believe `macros_replace` is a better name,, and we can certainly implement that.

betzw, 2016-10-25:
Actually, I do not care about the name.

betzw, 2016-11-14:
Any news on this @bogdanm?

bogdanm, 2016-11-14:
Cc @theotherjimmy

theotherjimmy, 2016-11-14:
This has not been high priority for me. Sorry, but I have not started on it yet.

betzw, 2016-12-27:
With the latest modifications to mbed-cli even my original approach (i.e. removing the macro with its original value and adding it again with the new value) does not work anymore. Could you pls. re-consider the proposal I have made here again?
Thanks in advance!

@theotherjimmy @screamerbg

RobMeades, 2017-04-12:
I also hit this issue recently when I wanted to change the value of a macro defined in `targets.json` from `mbed_app.json`.  I agree with @betzw that having to do, in my case:

```
            "target.macros_remove": ["HSE_VALUE=12000000"],
            "target.macros_add": ["HSE_VALUE=8000000"]
```
...is a bit cumbersome and a mechanism for simply changing the value would be a good improvement.  However, it is worth pointing out that the above did work for me (code as at start of March 2017 I think).

0xc0170, 2017-04-21:
Why don't we use config for this, using macros with values does not look right - thus the syntax add/remove as you have to use the very same definition =VALUE ?

theotherjimmy, 2017-04-21:
@0xc0170 I agree: we should use config instead.

betzw, 2017-04-21:
The only disadvantage I see in using `config` is that you cannot freely choose the macro's name. 
Don't think that this is a great issue, though.

theotherjimmy, 2017-04-21:
@betzw you can choose a portion of the name, so that should be Good Enough™

0xc0170, 2017-04-24:
Can we close this, as a resolution - use config? This is known limitation of using ``macros`` ?

closed 2017-04-26 by betzw