ARMmbed/mbed-os #3125: No email notifications

betzw, 2016-10-25:
### Description
- Type: Question
- Priority: Minor

-----------------------------------------------------------------

## Question

I have noticed in the last weeks that it seems as I do not get email notifications for the issues and pull requests authored by myself even if my settings should specify this.
Any idea?

cc @ARMmbed

0xc0170, 2016-10-25:
Seems to be working fine for me. You shall review your settings for This organization/repository or contact github.

betzw, 2016-10-26:
I have done both, but couldn't yet resolve the problem ...

0xc0170, 2016-10-28:
I am not familiar with any other settings for the organization. I don't think anyone can help you here besides github support.

adbridge, 2016-11-03:
@betzw 
On the notifications page of the settings can you confirm you have all the following ticked:
Participating:
email and web

Watching:
email and web

Choose which email updates you receive on conversations you're participating in or watching:

Comments on Issues and Pull Requests
Pull Request reviews
Pull Request pushes
Include your own updates 

and also check that you are actually watching the mbed-os repo, I couldn't see you on the watchers list when I just had a quick look?

closed 2016-11-05 by betzw

betzw, 2016-11-07:
Seems as if the problem has been resolved by GitHub support.
Thanks anyhow @adbridge for your questions.