ARMmbed/mbed-os #3126: Flash driver K64F (non flex memory) does not properly collaborate mbed5 OS.

EduardPon, 2016-10-25:
We would like to have a non volatile memory to store configuration data that must survive a reset/power cycle. So the using the HAL driver flash functions would be the way to go. But unfortunately these functions seem to block the OS too long, causing some issues in the OS. For example the erase function, it won't work without disabling the interrupts.
```
    __disable_irq();
    status = FLASH_Erase( &flash_config, 0xB3010, 16 , kFLASH_apiEraseKey );
    __enable_irq(); 
```

Also programming the flash can only be performed with 16 bytes at a time.

Please advise how to implement a decent non volatile memory solution. Thanks in advance.

0xc0170, 2016-10-25:
@rgrover

rgrover, 2016-10-25:
Would using the FAT filesystem work for you? It is an existing, mature solution as long as your application can work against an SDCard back-end for the filesystem.
Writing against an SDCard filesystem would not require blocking the OS for periods of time. The filesystem will also internally align arbitrary writes to the underlying SDCard block-size.
Could this work for you?

EduardPon, 2016-10-25:
It is not intended to use a SDCard. 
For the product we have in mind, we have to use the free part of the internal flash memory of the K64F.

rgrover, 2016-10-25:
If your application code can fit within the first 512K of internal flash, then the second 512K bank is free for storing data. In this case you can enjoy the impunity to use the second bank without blocking interrupts across FLASH erase/program operations. Interrupts need to blocked only if there is a possibility of ISRs running out of the same flash bank as data operations. Can you separate code and data into distinct flash banks?

EduardPon, 2016-10-25:
Yes must be feasible. 
So only accessing the second bank is sufficient, no flash configuration changes are required? 
Thanks.

rgrover, 2016-10-25:
if code and data are on distinct flash banks, then the default operation of flash is safe. No config changes are necessary. I believe the default K64F flash driver encourages and supports this separation. You should be fine just driving flash to write your data into bank-1 (leaving bank-0 for code).

EduardPon, 2016-10-25:
But if we run out and have to use also the second bank, what would than be a solution? Interrupt disabling would be the only option?

rgrover, 2016-10-25:
yes, the moment code crosses over into the second/'data' flash bank, you'll need to ensure that either:

* code for the flash driver is executing out of RAM or is located on bank-0 AND
* interrupts (in general) are blocked across flash operations or all potentially active ISRs execute code located on bank-0.

executing flash driver out of SRAM and disabling interrupts in general are the easiest way to accomplish the above.

EduardPon, 2016-10-25:
Clear, many thanks.

closed 2016-10-25 by EduardPon

EduardPon, 2016-10-31:
Hello, 
I still have a question. I'm using currently to disable the interrupts the call __disable_irq().
Presumably this is not safe and I have to use something like core_util_critical_section_enter()?
Thanks in advance.

rgrover, 2016-10-31:
@0xc0170 perhaps you can answer this

0xc0170, 2016-10-31:
>  Presumably this is not safe and I have to use something like core_util_critical_section_enter()?

That's correct, please use it.