ARMmbed/mbed-os #3130: Ignoring the return value of _lock.lock() in *Socket*

daniel-cesarini-tridonic-com, 2016-10-25:
### Description
The return value of _lock.lock() in *Socket* is ignored, leading to wrong software behavior. 

- Type: Bug
- Priority: Critical

---------------------------------------------------------------
## Bug

**Target**
All

**mbed-os sha:**
ba99a1f31d341f696764525abc05abccccb5e61a

**Expected behavior**
The return value of  ```_lock.lock();``` should be considered in all places of the Socket and UDPSocket and TCPSocket (e.g., constructor, destructor, send_to, recvfrom, ...). 

**Actual behavior**
The return value of ```_lock.lock();``` is never considered. The lock() function may return a value indicating that the lock on the mutex did not succeed (because the lower layer has been called with wrong parameters, it has been called from interrupt context, ...). 
In such a case (we experienced that a few times already) the mutual exclusion between the users of the Object is not granted anymore, and misbehavior of the software arises.

**Steps to reproduce**
Race condition, difficult to reproduce, but statically proven to be erroneous in the code.
----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
The problem should be solved, as it may lead to corrupted execution flows.

I did not still find a way of solving the issue in the Socket implementation, given the fact that several of the methods of the Socket return void and thus cannot communicate whether the operation has been successful or not.

@MarceloSalazar @karsev

0xc0170, 2016-10-25:
cc @geky @c1728p9

c1728p9, 2016-10-25:
Hi @daniel-cesarini-tridonic-com, this behavior is found in several other places in the codebase and is not specific to sockets. The currently planned solution is to treat mutex acquisitions in a critical section or interrupt to be a fatal error and trap on it. That way no error needs to be propagated in the return value. The change might be made slowly though to give time for code with this problem to be fixed.

daniel-cesarini-tridonic-com, 2016-10-27:
Hi @c1728p, you say that "The currently planned solution is to treat mutex acquisitions in a critical section or interrupt to be a fatal error and trap on it.". Could you please elaborate on this? 

In our case we had an Socket, containing the Mutex object, that was deleted during the error handling on the Network send function. Although that Socket had been deleted, a tasklet, scheduled with Nanostack events mechanism, tried to perform again a send, and finding the Socket deleted, with all its objects deleted too. Then the _lock.lock() failed, because of missing parameters. Would such a case also be covered by the "planned solution"?

daniel-cesarini-tridonic-com, 2017-11-29:
@c1728p9 @sg- are there any news on these issues / topics?

c1728p9, 2017-11-29:
Hi @daniel-cesarini-tridonic-com, trapping is enabled when you build as debug or set the macro MBED_TRAP_ERRORS_ENABLED=1. If you build in this configuration you'll be able to easily tell if anything is using a mutex from an interrupt handler.

daniel-cesarini-tridonic-com, 2017-11-30:
@c1728p9 yes, we are using MBED_TRAP_ERRORS_ENABLED=1, and that might solve part of the issues, but another issue that we found is partially related to this one is described in a MR in the PAL: https://github.com/ARMmbed/pal/pull/66.

Furthermore analyzing the lock mechanism (in the mbedOS Mutex class), that finally is using the RTX https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_mutex.c#L458 that over an SVC call is coming to https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_mutex.c#L162, I found out that actually the reasons for returning from it without having acquired the lock itself can be related also to other reasons than only being executed in ISR context (visible in https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_mutex.c#L458). 
Thus the lack of an evaluation of _lock.lock() method is still relevant and might possibly lead to SERIOUS errors in the software, IMHO!

FYI: @sg- @MarceloSalazar @karsev

c1728p9, 2017-11-30:
Hi  @daniel-cesarini-tridonic-com I created #5624 as a potential fix for this issue.

daniel-cesarini-tridonic-com, 2017-12-01:
Hi @c1728p9 , I checked that, thanks. 
Still work to be done, as it looks like.

MarceloSalazar, 2017-12-13:
@c1728p9 given that https://github.com/ARMmbed/mbed-os/pull/5624 is closed, what do you propose?

c1728p9, 2017-12-13:
I think the tentative plan is to make mutex lock failures (other than timeout) fatal in debug and development builds for mbed-os 5.8. @sg- needs to confirm this though.

daniel-cesarini-tridonic-com, 2017-12-14:
@c1728p9 @sg- what does this mean? 

> plan is to make mutex lock failures (other than timeout) fatal in debug and development builds

Furthermore, in general, should mutex / other concurrency primitives have a return value at all? 
And if yes, would that mean that *all* users of such primitives should handle the possible return values?

c1728p9, 2017-12-14:
@daniel-cesarini-tridonic-com the return values can't be removed in mbed 5 as this would break backwards compatibility, otherwise the return values would be removed. Errors that indicate a bug in users code, such as a mutex lock failing, will be fatal in debug and develop builds so the error cannot be ignored.

daniel-cesarini-tridonic-com, 2017-12-15:
@c1728p9 : and what should happen in production builds? How should (possible) errors be handled there?
Should they silently happen ... without any notice?

c1728p9, 2017-12-15:
@daniel-cesarini-tridonic-com runtime errors will still cause a trap by the use of the `error` function. The sanity checking asserts are turned off and ignored. This is not my call to make. If you are unhappy with it then please open an issue to discuss this.

As for this ticket, it sounds like the consensus is socket shouldn't need to check the return value of lock. Unless you have more questions/concerns specifically related to sockets then this issue should be closed.

CC: @geky, @kjbracey-arm, @sg-

daniel-cesarini-tridonic-com, 2017-12-18:
@c1728p9 : I agree this issue should / can be closed IMHO.
Only one question left: 

> runtime errors will still cause a trap by the use of the error function
where would that happen? Should it be in the context of the mutex class itself? Is it already in place / planned to be implemented? Is there another MR / Issue for tracking the mutex implementation?

c1728p9, 2017-12-18:
Where this is implemented is to be determined. It could be in the Mutex class or in the RTX handlers. Also, to be determined is if incorrect use of a mutex is fatal in release builds. Again, this is not my call to make and should be discussed in a separate issue.  I don't think there is an issue for this discussion, so feel free to create one.

daniel-cesarini-tridonic-com, 2018-05-11:
Opened new issue https://github.com/ARMmbed/mbed-os/issues/6872 as suggested by https://github.com/c1728p9 in https://github.com/ARMmbed/mbed-os/issues/3130#issuecomment-352028845 and https://github.com/ARMmbed/mbed-os/issues/3130#issuecomment-352493443

daniel-cesarini-tridonic-com, 2018-05-11:
I will now close this issue and follow on the discussion in the newly created one.

closed 2018-05-11 by daniel-cesarini-tridonic-com