ARMmbed/mbed-os #3131: Runtime dynamic memory tracing always returns malloc and free as callers

javier-moreno-tridonic-com, 2016-10-25:
### Description
- Type: Bug 
- Related PR: https://github.com/ARMmbed/mbed-os/pull/2487/commits/8b908abcc521c72b8ce3cfbd467ee40a950a28a4 @bogdanm 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NucleoF091

**Toolchain:**
GCC_ARM arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release)

**Actual behaviour**
All the calls made to **MBED_CALLER_ADDR()** inside __wrap__malloc_r and __wrap__free_r will return the addresses were malloc and free are located and not who called malloc or free

**Steps to reproduce**
I just run and mbed example with compiled with the option **-DMBED_MEM_TRACING_ENABLED=1**

```
Stack trace content

__wrap__malloc_r() at mbed_alloc_wrappers.cpp:104 0x800f168	
malloc() at 0x80164dc	
m2m::String::strdup() at m2mstring.cpp:29 0x8008506	
m2m::String::String() at m2mstring.cpp:60 0x800856c	
MbedClient::create_interface() at simpleclient.h:114 0x80006ac	
main() at main.cpp:134 0x8000914	

```

> log output **#m:0x200050c8;0x80164dd-14**

Which means that the caller is equal to 0x80164dd (close to 0x80164dc)

The weirdest thing for me is that as the stack indicates **malloc** is "calling" **__wrap__malloc_r** which doesn't make any sense to me. Specially since **__wrap__malloc_r** is calling ** ptr = __real__malloc_r(r, size);** which is supposed to point to the actual malloc function. 

Compiler optimisations going wild?

bogdanm, 2016-10-25:
I'm unable to reproduce this issue with this modified blinky code:

```
#include <stdio.h>
#include <stdlib.h>
#include "mbed.h"
#include "rtos.h"
#include "mbed_mem_trace.h"

DigitalOut led1(LED1);

// main() runs in its own thread in the OS
// (note the calls to Thread::wait below for delays)
int main() {
    mbed_mem_trace_set_callback(mbed_mem_trace_default_callback);
    while (true) {
        void *p = malloc(50);
        led1 = !led1;
        Thread::wait(500);
        free(p);
    }
}
```

and `MBED_MEM_TRACING_ENABLED` defined, on a very recent SHA1 of mbed-os. I'm getting the proper addresses in the tracer's output. Note, however, that `MBED_OS_CALLER_ADDR()` is not necessarily reliable. @c1728p9 made a suggestion to improve this situation a bit, I'll try to do that.

About the other part (the stack trace), I'm guessing it has something to do with the magic behind ld's `--wrap-symbol` option. Again, should probably be taken with a bit of salt :)

sg-, 2017-02-03:
@javier-moreno-tridonic-com @bogdanm Any further investigation needed here or can it be closed?

javier-moreno-tridonic-com, 2017-02-06:
@sg- Well to be honest I did not get the chance to dig any further on this topic. As far as I'm concerned can be closed.

I experienced this issue with one of the initial releases of mbed 5. If it persists with the current one I can always reopen it.

closed 2017-02-06 by 0xc0170