ARMmbed/mbed-os #3137: Re-connecting to mesh network after calling `disconnect()` fails.

tommikas, 2016-10-26:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

`connect()` can only be called once on a mesh interface, even after calling `disconnect()`.

The problem seems to be due to https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/nanostack-interface/NanostackInterface.cpp#L596 trying to initialize things every time `connect()` is called, even if it was already done.

**Expected behavior**

`connect()` can be called after `disconnect()`

**Actual behavior**

The second connect() fails with NSAPI_ERROR_DEVICE_ERROR

**Steps to reproduce**

1. Create a new mesh interface.
2. Call `connect()`, connecting to the network
3. Call `disconnect()`
4. Call `connect()` again.

0xc0170, 2016-10-28:
@SeppoTakalo @kjbracey-arm

sg-, 2017-02-03:
@tommikas Has this been resolved?

tommikas, 2017-02-06:
No, it still doesn't work.

@SeppoTakalo @kjbracey-arm Is this ever going to be possible? I personally don't care too much either way, but this is inconsistent with how other network stacks behave.

kjbracey-arm, 2017-02-06:
It's certainly possible, but pretty low priority - shutdown seems one of those things that seems like more a feature for tests, rather than a feature that is required for customers and needs to be tested.

Do you need this to work? Would disconnect and reconnect just being no-ops be sufficient?

tommikas, 2017-02-06:
I don't exactly need it. We have a test in place that checks that multiple connect-disconnect cycles work (because once upon a time it wasn't working for LWIP) but for mesh we just do connect-disconnect once, which works fine.

I'm fine with it being as it is if you don't mind the difference in behaviour between the different stacks. It should probably be documented and perhaps even explicitly handled though. It doesn't seem unreasonable to assume that connect can be called after disconnect, and it currently fails with an error() call, instead of for example returning an error value, because it's doing one-time initialization unconditionally in connect.

closed 2017-10-27 by ChiefBureaucraticOfficer