ARMmbed/mbed-os #3138: k64f_emac.c calls malloc from within IRQ context

bremoran, 2016-10-26:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Minor
- Related issue: #3118 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
All

**Toolchain version:**
All
**mbed-cli version:** 
All

**meed-os sha:**
e435a07d9252f133ea3d9f6c95dfb176f32ab9b6

**DAPLink version:**
N/A

**Expected behavior**
Memory allocation in IRQ context should be from pools only.

**Actual behavior**
Memory allocation in ethernet IRQ context can be from the heap. This can result in undesirable behaviour and poor network performance.

```C
pbuf_alloc(... , PBUF_RAM); // calls mem_malloc(), which appears to be using LwIP's own internal heap
```

**Steps to reproduce**
Source inspection

**Suggested fix**
In https://github.com/ARMmbed/mbed-os/blob/646cc89a620f978ed6174cb7bbb921e0f0d910f2/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c:

Replace https://github.com/ARMmbed/mbed-os/blob/646cc89a620f978ed6174cb7bbb921e0f0d910f2/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c#L512-L520
with:
```C
psend = memp_malloc(MEMP_PBUF_POOL);
```

also replace https://github.com/ARMmbed/mbed-os/blob/646cc89a620f978ed6174cb7bbb921e0f0d910f2/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c#L113 with 

```C
memp_free(MEMP_PBUF_POOL, tx_buff[i]);
```

There is also a `pbuf_alloc` in `k64f_low_level_input`, but removing that IRQ-context malloc will require additional investigation.

0xc0170, 2016-10-26:
cc @geky @bogdanm

infinnovation, 2016-10-26:
These are not in IRQ context - they are in separate threads.  The threads are scheduled from IRQ handlers via semaphores.

bremoran, 2016-10-26:
Based on the LwIP implementation guidelines, data should be queued in the IRQ handler, then processed in thread context.

bremoran, 2016-10-26:
For example, here is the sample driver that LwIP provides. Note that it allocates from `PBUF_POOL` rather than `PBUF_RAM`

https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/lwip_ethernetif.c#L200

geky, 2016-10-27:
If I understand correctly, this boils down to a tradeoff between interrupt latency and potentially increased packet loss if lwip can't get to the buffered packets in time. From a pragmatic standpoint, I'm not sure changing the existing driver is worth potential gains in this tradeoff. Do you have an example where the current strategy might break down?

Although, I don't think we'd be opposed to merging a patch that adopts lwip's suggested strategy (after significant testing of course).

@kjbracey-arm thoughts?

kjbracey-arm, 2016-10-28:
Our approach for Nanostack drivers in mbed OS 5 has been to do minimal work from interrupts - just signal, basically, and do everything in a real-time priority thread. Even the tightest interrupt timing requirements we have work fine like that. Keeps things generally simpler and more flexible - we're assuming an RTOS, so we may as well get some use out of it.

For an Ethernet driver with DMA like this, I don't see that IRQ versus thread makes much difference either way. There's not much driver processing to do compared to the line rate. The Nanostack thready version manages to keep up with the fastest flood ping we can throw at it, and I don't see why this wouldn't. Might be worth checking that your RX thread is higher priority than normal though.

Regardless you should probably be claiming from the pool anyway  - the LWIP pool is there and intended for use by drivers. It's presumably been sized on the assumption that the Ethernet is going to use it, and by not using it you're wasting that already-paid-for allocation. (The Nanostack driver does claim memory from Nanostack's ns_dyn_mem_lib "pool").

(My other beef with the lwIP K64F driver is that it wastes RAM by using 3 threads to do 3 tasks, rather than doing them all in 1 thread like the Nanostack one. And even that would be better using a system-wide realtime-priority event loop - do we have one declared yet?)

bremoran, 2016-10-28:
To further what @kjbracey-arm has said, allocating out of heap on receive is dangerous. On busy networks, it could easily lead to memory fragmentation, failed memory allocations, etc. This effectively opens a window to denial of service attacks against the MCU. Networks are lossy. It's part of network programming. Dropping a packet because the receive queue is full is much safer than allocating a buffer every time a packet arrives.

It's definitely reasonable to defer packet processing to thread context; I just think it should use the pool allocator, which is intended for that purpose.

bremoran, 2016-10-28:
Thinking about it further, by generating the right network traffic, it is possible to deliberately fragment a device's heap. Replacing malloc on receive with pool allocate on receive is not adequate on its own. If a remote device can influence the size of data transmitted by the MCU, then it's possible to cause memory fragmentation on send as well. Sufficiently fragmented memory has the potential to stop a device from operating.

geky, 2016-10-28:
This has been a very interesting discussion, and highlights several concerns with the existing driver (drivers?). I'll look into see how we can improve the threading impact and avoid heap fragmentation.

@bremoran, do you think you could put together what network traffic would cause this sort of fragmentation? This sort of attack would be a great thing to add to tests to catch this in other drivers.

closed 2017-10-27 by ChiefBureaucraticOfficer