ARMmbed/mbed-os #3145: Base DNS code defaults to type A queries on IPv6 interfaces

kjbracey-arm, 2016-10-27:
- Type: Bug
- Priority: Major 

If the base DNS code is in use, eg for Nanostack rather than lwIP, no attempt is made to perform the correct type of address query by default. SocketAddress(iface, "ntp.pool.org"), as seen in the CLI test app, will perform an IPv4 A type lookup when using Nanostack and iface is IPv6 only.

At some point in the lwIP IPv6 development we had the top level gethostbyname() do a check for iface->get_ip_address() containing a ':', and requesting IPv6 if so.  This fell out after one of the NetworkStack/NetworkInterface shuffles caused a problem (which I do not recall).

The lwIP implementation has a similar "A or AAAA based on indicated interface address" heuristic built-in - it would be redundant if done at the generic layers.

kjbracey-arm, 2016-10-27:
@SeppoTakalo, @mikaleppanen, @c1728p9, @geky

geky, 2016-10-27:
Interesting, my mistake on missing the ':' heuristic entirely.

The current design does allow IPv6 network interfaces to default to AAAA queries if the interface overloads the `gethostbyname` function (which can just call `nsapi_dns_query`).

But I your ':' heuristic makes a lot of sense and should be easy to adopt.

kjbracey-arm, 2016-10-28:
It is a slightly sneaky heuristic, but it takes advantage of the fact we have this "get_ip_address" API that reports a single address. It also handles any preference in a dual-stack type scenario - if an interface does have both IPv4 and IPv6, it will presumably be reporting its preferred one via that call.

Hadn't occurred to me to do the custom overload redirecting to nsapi_dns_query.

closed 2016-11-01 by sg-