ARMmbed/mbed-os #3148: KL25Z test regression - Basic thread for mbed-os 2

adbridge, 2016-10-27:
### Description

basic_thread test for the mbed-os-2 test suite failed.

- Type: Bug
- Priority: Major 

## Bug

**Target**
FL25Z
**Toolchain:**
GCC_ARM

**Toolchain version:**
Latest
**mbed-cli version:** 
Latest
**meed-os sha:**
mbed-os-5.2.1 candidate 

**Expected behavior**

MBED: Instrumentation: "COM55" and disk: "N:"
HOST: Copy image onto target...
        1 file(s) copied.
Looking for KL25Z with MBEDLS
Remount complete
HOST: Initialize serial port...
Looking for KL25Z with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default_auto'
HOST: Property 'description' = 'Signals messaging'
HOST: Property 'test_id' = 'RTOS_4'
HOST: Start test...
Handled 100 signals
{{success}}
{{end}}

**Actual behavior**

MBED: Instrumentation: "COM55" and disk: "N:"
HOST: Copy image onto target...
        1 file(s) copied.
Looking for KL25Z with MBEDLS
Remount complete
HOST: Initialize serial port...
Looking for KL25Z with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default_auto'
HOST: Property 'description' = 'Signals messaging'
HOST: Property 'test_id' = 'RTOS_4'
HOST: Start test...
Operator new[] out of memory

0xc0170, 2016-10-28:
@mmahadevan108

mmahadevan108, 2016-11-14:
@maclobdell

mmahadevan108, 2016-12-14:
I ran the test suite against the latest code-base, the tests passed.
Test summary:
+---------------+--------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| Result        | Target | Toolchain | Test ID     | Test Description                      | Elapsed Time (sec) | Timeout (sec) | Loops |
+---------------+--------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| OK            | KL25Z  | ARM       | DTCT_1      | Simple detect test                    |        0.5         |       10      |  1/1  |
| OK            | KL25Z  | ARM       | EXAMPLE_1   | /dev/null                             |        3.47        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | KL25Z_5     | MMA8451Q accelerometer                |       12.53        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_10     | Hello World                           |        0.4         |       5       |  1/1  |
| OK            | KL25Z  | ARM       | MBED_11     | Ticker Int                            |       11.38        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_12     | C++                                   |        1.36        |       10      |  1/1  |
| NOT_SUPPORTED | KL25Z  | ARM       | MBED_16     | RTC                                   |         0          |       0       |   -   |
| OK            | KL25Z  | ARM       | MBED_2      | stdio                                 |        0.74        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_23     | Ticker Int us                         |        11.4        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_24     | Timeout Int us                        |       11.66        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_25     | Time us                               |       11.36        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_26     | Integer constant division             |        1.36        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_34     | Ticker Two callbacks                  |       11.38        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_37     | Serial NC RX                          |        6.96        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_38     | Serial NC TX                          |        5.94        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_A1     | Basic                                 |        1.37        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_A21    | Call function before main (mbed_main) |        1.46        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_A9     | Serial Echo at 115200                 |        1.3         |       20      |  1/1  |
| OK            | KL25Z  | ARM       | MBED_BUSOUT | BusOut                                |        2.27        |       30      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_1      | Basic thread                          |       11.38        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_2      | Mutex resource lock                   |        11.5        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_3      | Semaphore resource lock               |        8.48        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_4      | Signals messaging                     |        6.49        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_5      | Queue messaging                       |        2.41        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_6      | Mail messaging                        |        2.41        |       20      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_7      | Timer                                 |        11.4        |       15      |  1/1  |
| OK            | KL25Z  | ARM       | RTOS_8      | ISR (Queue)                           |        6.46        |       20      |  1/1  |
+---------------+--------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
Result: 26 OK / 1 NOT_SUPPORTED

Completed in 378.36 sec

theotherjimmy, 2016-12-20:
@adbridge Looks like we fixed this in the mean time.

mmahadevan108, 2017-01-12:
Can we close this issue?

bridadan, 2017-01-12:
@mmahadevan108 I noticed the original issue was reported to be a problem when using the GCC_ARM toolchain, not the ARM toolchain. Would you mind confirming that the issue is resolved with the GCC_ARM toolchain as well?

mmahadevan108, 2017-01-12:
Thanks Brian. I reran the tests using GCC_ARM and I see failures

mmahadevan108, 2017-01-12:
We are probably overflowing the heap and/or stack. Not sure if there is a fix, please suggest.

bridadan, 2017-01-12:
@mmahadevan108 You can adjust the stack size like the rest of the targets in those tests.

closed 2017-10-27 by ChiefBureaucraticOfficer