ARMmbed/mbed-os #3149: KL25Z test regression - basic thread for mbed-os 2

adbridge, 2016-10-27:
### Description

basic thread test from mbed-os-2 test suite is failing.

- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
KL25Z

**Toolchain:**
GCC_ARM

**Toolchain version:**
latest
**mbed-cli version:** 
latest
**meed-os sha:**
mbed-os-5.2.1 release candidate

**Expected behavior**

MBED: Instrumentation: "COM55" and disk: "N:"
HOST: Copy image onto target...
        1 file(s) copied.
Looking for KL25Z with MBEDLS
Looking for KL25Z with MBEDLS
Remount complete
HOST: Initialize serial port...
Looking for KL25Z with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '15'
HOST: Property 'host_test_name' = 'wait_us_auto'
HOST: Property 'description' = 'Basic thread'
HOST: Property 'test_id' = 'RTOS_1'
HOST: Start test...
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
* in 1.00 sec (0.00) [OK]
Consecutive OK timer reads: 10
Completed in 10.00 sec

{{success}}
{{end}}

**Actual behavior**

MBED: Instrumentation: "COM55" and disk: "N:"
HOST: Copy image onto target...
        1 file(s) copied.
Looking for KL25Z with MBEDLS
Remount complete
HOST: Initialize serial port...
Looking for KL25Z with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '15'
HOST: Property 'host_test_name' = 'wait_us_auto'
HOST: Property 'description' = 'Basic thread'
HOST: Property 'test_id' = 'RTOS_1'
HOST: Start test...
  in 0.00 sec (1.00) [FAIL]
  in 0.00 sec (1.00) [FAIL]
  in 0.00 sec (1.00) [FAIL]
  in 0.00 sec (1.00) [FAIL]
  in 0.00 sec (1.00) [FAIL]
n in 0.00 sec (1.00) [FAIL]
e in 0.00 sec (1.00) [FAIL]
w in 0.00 sec (1.00) [FAIL]
[ in 0.00 sec (1.00) [FAIL]
] in 0.00 sec (1.00) [FAIL]
  in 0.00 sec (1.00) [FAIL]
o in 0.00 sec (1.00) [FAIL]
u in 0.00 sec (1.00) [FAIL]
Consecutive OK timer reads: 0
Completed in 0.01 sec

{{failure}}
{{end}}

0xc0170, 2016-10-28:
cc @mmahadevan108

mmahadevan108, 2016-11-14:
cc @maclobdell

maclobdell, 2016-11-30:
@mmahadevan108  This test can be recreated by running the old test system.  I had to refresh myself on how to do it.  It is very simple to do.  Here are the steps.  
From within the mbed-os/tools/ directory, create muts_all.json and test_spec.json files as described here:  https://developer.mbed.org/teams/SDK-Development/wiki/Mbed-SDK-automated-test-suite
Then execute the command to run the test:
singletest.py -i test_spec.json -M muts_all.json -n RTOS_1 -v

The elf file that can be debugged will get generated here:
mbed-os\BUILD\test\KL25Z\GCC_ARM\RTOS_1

I suspect that there is a memory configuration issue that could have caused this.  

@c1728p9  do you have any suggestions?

c1728p9, 2016-12-12:
It looks like the KL25Z is running out of memory. When I run RTOS_1 I get this output:
```
{{timeout;15}}
{{host_test_name;wait_us_auto}}
{{description;Basic thread}}
{{test_id;RTOS_1}}
{{start}}
Operator new[] out of memory
```

mmahadevan108, 2016-12-14:
Did KL25Z pass mbed tests before?

mmahadevan108, 2016-12-14:
I ran these tested against the latest code and the tests passed:
OK            | KL25Z  | ARM       | RTOS_1      | Basic thread                          |       11.38        |       15      |  1/1  |

theotherjimmy, 2016-12-20:
@adbridge If these tests now pass can we close this issue?

mmahadevan108, 2017-01-12:
Can we close this issue?

bridadan, 2017-01-12:
@mmahadevan108 I noticed the original issue was reported to be a problem when using the GCC_ARM toolchain, not the ARM toolchain. Would you mind confirming that the issue is resolved with the GCC_ARM toolchain as well?

mmahadevan108, 2017-01-12:
Thanks Brian. I re-ran using GCC_ARM and still see failures.

mmahadevan108, 2017-01-12:
We are probably overflowing the heap and/or stack. Not sure if there is a fix, please suggest.

closed 2017-10-27 by ChiefBureaucraticOfficer