ARMmbed/mbed-os #3152: Odin WiFi broken by DNS API change

bulislaw, 2016-10-27:
This change `1f4eb0aaa1499f8de88d10c5987af60c2395c9da` by @geky broke compatibility of prebuilt WiFi library for Odin board.

```
Build failures:
  * UBLOX_EVK_ODIN_W2::GCC_ARM::TESTS-NETWORK-WIFI
        Building project wifi (UBLOX_EVK_ODIN_W2, GCC_ARM)
        Scan: GCC_ARM
        Scan: FEATURE_LWIP
        Scan: wifi
        Compile [100.0%]: main.cpp
        [Warning] cc.h@57,1: "BYTE_ORDER" redefined
        Link: wifi
        /Users/barsza01/devel/mbed/code/mbed-os/BUILD/tests/UBLOX_EVK_ODIN_W2/GCC_ARM/targets/TARGET_STM/TARGET_STM32F4/TARGET_UBLOX_EVK_ODIN_W2/sdk/TOOLCHAIN_GCC_ARM/libublox-odin-w2-driver.a(OdinWiFiInterface.o):(.rodata._ZTV17OdinWiFiInterface+0x30): undefined reference to `NetworkInterface::gethostbyname(char const*, SocketAddress*)'
        collect2: error: ld returned 1 exit status
```

@andreaslarssonublox could you rebase and rebuild the libraries please.

0xc0170, 2016-10-28:
Thanks @bulislaw for reporting this. Why the CI did not report this, did we miss something?

> This change 1f4eb0aaa1499f8de88d10c5987af60c2395c9da by @geky broke compatibility of prebuilt WiFi library for Odin board.

We need to be careful with this type of changes. Note taken.

bulislaw, 2016-10-28:
Yes :) We're missing WiFi tests and the EMAC is disabled by default. https://github.com/ARMmbed/mbed-os/pull/3147

geky, 2016-10-28:
Changes like those are always a minefield, sorry for the trouble.

bulislaw, 2016-10-31:
Yeah, especially with a binaries in the sources, that's why I want to have tests for Odin. #3147

0xc0170, 2016-11-09:
Fixed via #3221

closed 2016-11-09 by 0xc0170