ARMmbed/mbed-os #3153: FRDM-KL43Z Test Failures

sg-, 2016-10-27:
```
python tools/singletest.py --auto --tc GCC_ARM --report-build=test_build_report.xml --report-junit=test_run_report.xml --waterfall --consolidate-waterfall --global-loops=3 -V

...
| FAIL   | KL43Z[9900] | GCC_ARM   | MBED_38     | Serial NC TX                          |        0.91        |       20      |  0/3  |
```

Serial NC test fails with ARM, GCC_ARM and IAR

[KL43Z_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557026/KL43Z_ARM_singletest.txt)
[KL43Z_GCC_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557028/KL43Z_GCC_ARM_singletest.txt)
[KL43Z_IAR_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557027/KL43Z_IAR_singletest.txt)

0xc0170, 2016-10-28:
@mmahadevan108

mmahadevan108, 2016-10-28:
I had reported this issue sometime ago when submitting the pull request for KL43Z support and had submitted the below PR as fix: https://github.com/ARMmbed/mbed-os/pull/2243

Since the free function is not invoked the UART driver's deinit function is not called. Adding this change to SerialBase.* files will fix this issue.

mmahadevan108, 2016-11-14:
@maclobdell

closed 2017-10-27 by ChiefBureaucraticOfficer