ARMmbed/mbed-os #3154: FRDM-KL46Z test failures

sg-, 2016-10-27:
```
python tools/singletest.py --auto --tc GCC_ARM --report-build=test_build_report.xml --report-junit=test_run_report.xml --waterfall --consolidate-waterfall --global-loops=3 -V
...
| FAIL   | KL46Z[9900] | GCC_ARM   | MBED_16     | RTC                                   |        4.6         |       20      |  0/3  |

```

Fails with ARM, GCC_ARM and IAR

[KL46Z_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557039/KL46Z_ARM_singletest.txt)
[KL46Z_GCC_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557038/KL46Z_GCC_ARM_singletest.txt)
[KL46Z_IAR_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557040/KL46Z_IAR_singletest.txt)

0xc0170, 2016-10-28:
@mmahadevan108

mmahadevan108, 2016-11-14:
cc @maclobdell

closed 2017-10-27 by ChiefBureaucraticOfficer