ARMmbed/mbed-os #3155: FRDM-KL05 test failures

sg-, 2016-10-27:
```
python tools/singletest.py --auto --tc GCC_ARM --report-build=test_build_report.xml --report-junit=test_run_report.xml --waterfall --consolidate-waterfall --global-loops=3 -V
...
| FAIL          | KL05Z[9900] | GCC_ARM   | MBED_37     | Serial NC RX                          |        2.71        |       20      |  0/3  |
| FAIL          | KL05Z[9900] | GCC_ARM   | MBED_38     | Serial NC TX                          |        1.72        |       20      |  0/3  |
```

Fails with IAR, ARM and GCC_ARM

[KL05Z_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557050/KL05Z_ARM_singletest.txt)
[KL05Z_GCC_ARM_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557048/KL05Z_GCC_ARM_singletest.txt)
[KL05Z_IAR_singletest.txt](https://github.com/ARMmbed/mbed-os/files/557049/KL05Z_IAR_singletest.txt)

0xc0170, 2016-10-28:
@mmahadevan108

mmahadevan108, 2016-11-14:
cc @maclobdell

closed 2017-10-27 by ChiefBureaucraticOfficer