ARMmbed/mbed-os #3164: UART Serial printf not working while compiled via GCC_ARM. NUCLEO_F767ZI

sergiuszm, 2016-10-30:
### Description
- Type: Bug
- Priority: Blocker

I am unable to print to serial output using UART. 

I have used examples from [mbed/SerialPC](https://developer.mbed.org/handbook/SerialPC). 

To confirm that I didn't make a mistake while using SerialPC examples I have compiled the same code via GCC_ARM and via mbed online compiler (it uses ARM Compiler if I am right). The code compiled via GCC_AMR doesn't print anything or the same strange string (`$$$081802210D2F651E326BFC3B`) while code compiled via mbed online compiler works as expected.

I have checked tests from this repository and there is one which is trying to get output from a serial port ` python tools/make.py -m NUCLEO_F767ZI -t GCC_ARM -d /media/serchio/NODE_F767ZI/ -s /dev/ttyACM0 -p 0`. I didn't get a correct results because execution just freeze.

Result in MacOS:
`Building project basic (NUCLEO_F767ZI, GCC_ARM)
Scan: basic
Scan: mbed
Scan: env
Compile [ 50.0%]: main.cpp
Compile [100.0%]: test_env.cpp
Link: basic
Elf2Bin: basic
+-------------+-------+-------+------+
| Module      | .text | .data | .bss |
+-------------+-------+-------+------+
| .build/mbed |  8461 |     8 | 1440 |
| .build/test |   599 |    16 |    0 |
| Fill        |    53 |     0 |    3 |
| Misc        | 27088 |  2484 |   97 |
| Subtotals   | 36201 |  2508 | 1540 |
+-------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 4048 bytes
Total RAM memory (data + bss + heap + stack): 4048 bytes
Total Flash memory (text + data + misc): 38709 bytes

Image: /Users/serchio/Documents/Projects/mbed-os/.build/test/NUCLEO_F767ZI/GCC_ARM/MBED_A1/basic.bin
$$$081802210D2F651E326BFC3B{{timeout;20}}
{{host_test_name;default_auto}}
{{description;Basic}}
{{test_id;MBED_A1}}
{{start}}
{{success}}
{{end}}
^C
[CTRL+c] exit`
 
Result in Linux:
`Building project basic (NUCLEO_F767ZI, GCC_ARM)
Scan: basic
Scan: mbed
Scan: env
+-------------+-------+-------+------+
| Module      | .text | .data | .bss |
+-------------+-------+-------+------+
| .build/mbed |  8431 |     8 | 1440 |
| .build/test |   589 |    16 |    0 |
| Fill        |    47 |     0 |    3 |
| Misc        | 27068 |  2484 |   97 |
| Subtotals   | 36135 |  2508 | 1540 |
+-------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 4048 bytes
Total RAM memory (data + bss + heap + stack): 4048 bytes
Total Flash memory (text + data + misc): 38643 bytes

Image: /home/serchio/Documents/mbed-os/.build/test/NUCLEO_F767ZI/GCC_ARM/MBED_A1/basic.bin
{{timeout;20}}
{{host_test_name;default_auto}}
{{description;Basic}}
{{test_id;MBED_A1}}
{{start}}
{{success}}
{{end}}
^C
[CTRL+c] exit`

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F767ZI

**Toolchain:**
GCC_ARM

**Toolchain version:**
MacOs: version 5.4.1 20160919
Linux Mint: version 5.4.1 20160919

**meed-os sha:**
c9f9ffa

**Expected behavior**
Get output from the serial port using `Serial pc.printf()`

**Actual behavior**
No output or the same strange string: `$$$081802210D2F651E326BFC3B`

**Steps to reproduce**
` python tools/make.py -m NUCLEO_F767ZI -t GCC_ARM -d /media/serchio/NODE_F767ZI/ -s /dev/ttyACM0 -p 0`

----------------------------------------------------------------

SWO printing is working as expected. The problem is probably connected to UART (which is used by `Serial pc.printf()`?).

0xc0170, 2016-10-31:
@bcostm @adustm @LMESTM

bridadan, 2016-10-31:
@sergiuszm If you haven't already done so, it'd be worth upgrading the version of ST-Link on the board: http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link007.html

sergiuszm, 2016-10-31:
@bridadan Thanks, I have upgraded my firmware from V2J27M15 to V2J28M16. Unfortunately, it didn't fix a problem.

`Building project basic (NUCLEO_F767ZI, GCC_ARM)
Scan: basic
Scan: mbed
Scan: env
Elf2Bin: basic
+-------------+-------+-------+------+
| Module      | .text | .data | .bss |
+-------------+-------+-------+------+
| .build/mbed |  8461 |     8 | 1440 |
| .build/test |   599 |    16 |    0 |
| Fill        |    53 |     0 |    3 |
| Misc        | 27088 |  2484 |   97 |
| Subtotals   | 36201 |  2508 | 1540 |
+-------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 4048 bytes
Total RAM memory (data + bss + heap + stack): 4048 bytes
Total Flash memory (text + data + misc): 38709 bytes

Image: /Users/serchio/Documents/Projects/mbed-os/.build/test/NUCLEO_F767ZI/GCC_ARM/MBED_A1/basic.bin
$$$081802210D2F651E326BFC3B{{timeout;20}}
{{host_test_name;default_auto}}
{{description;Basic}}
{{test_id;MBED_A1}}
{{start}}
{{success}}
{{end}}`

bridadan, 2016-10-31:
@sergiuszm Thanks for trying that! At this point I'll hand it off to ST :smile:

bcostm, 2016-11-01:
If I understand well you have an issue with MacOS but it is ok with Linux, correct ?
If yes, can you please change the title of this issue and add "with MacOS" ?

sergiuszm, 2016-11-01:
@bcostm Unfortunately, not. I have a problem with both platforms. I can't get a serial output - used test case was just an example of such situation.

This is another code that can be used to test the issue.
```
#include <mbed.h>

Serial pc(USBTX, USBRX);

int main() {
    pc.printf("Hello World!\n");
}
```

I should get a `Hello World!` on the serial output but I don't get anything.

The same code works perfectly on Nucleo L476RG or works if was compiled with ARM Compiler 5 instead of GNU_ARM.

bcostm, 2016-11-03:
Can you try to add a while(1) loop after the line pc.printf ?

bcostm, 2016-11-24:
Same behavior as in Issue #3223 
i.e.
- Exported Nucleo_printf example to uVision5.
- pc.printf does not print anything. I suspect also a hardfault but not checked
- if printf is used it is ok

sergiuszm, 2016-11-25:
@bcostm I can confirm that printf is working as expected and the problem is with pc.printf.

LMESTM, 2017-05-24:
@sergiuszm sorry for not coming back to you earlier. 
I just run the test of your example from Nov 1, with GCC_ARM on NUCLEO_F767ZI and I cannot reproduce the issue. (I'm on windows though ...)

f...0 MINGW64 /c/Data/Workspace/mbed ((c9f9ffa...))
$ python tools/make.py -m NUCLEO_F767ZI -t GCC_ARM -d /d/ -s COM66 -p 0
Building project basic (NUCLEO_F767ZI, GCC_ARM)
Scan: basic
Scan: mbed
Scan: env
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Fill      |    77 |     4 |    3 |
| Misc      | 53750 |  2500 | 1801 |
| Subtotals | 53827 |  2504 | 1804 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 4308 bytes
Total RAM memory (data + bss + heap + stack): 4308 bytes
Total Flash memory (text + data + misc): 56331 bytes

Image: C:\Data\Workspace\mbed\.build\test\NUCLEO_F767ZI\GCC_ARM\MBED_A1\basic.bin
Hello World!

The Hello World! output is just fine - I can't reproduce the issue you mention.

I'm using :
gcc version 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496] (GNU Tools for ARM Embedded Processors)

LMESTM, 2017-06-13:
@sergiuszm  I know it's an old bug, but I would not be able to help further without more information. Otherwise I'll consider this can be closed as I could successfully run the suggested test
ST_TO_BE_CLOSED

sergiuszm, 2017-06-13:
@LMESTM Ah, I'm sorry. I'm closing the issue since I am unable to check if it is resolved or not.

closed 2017-06-13 by sergiuszm

LMESTM, 2017-06-13:
@sergiuszm No pb - thanks for the update !