ARMmbed/mbed-os #3167: CFSTORE fails to return error code when write operations fail ("exhausted flash")

simonqhughes, 2016-10-31:
From: Jenia Kogan 
Sent: Friday, October 28, 2016 12:35 PM
To: Rohit Grover
Cc: Nimrod Zimerman; Simon Hughes; Milosch Meriac
Subject: Cfstore 

Hi Rohit,

We have some issue with cfstore.
We are using mbed-os 5.2 version and we have board with exhausted flash. Write operations to flash are failed on this board.
What is weird that we don't receive any indication about this failure from the cfstore driver.

Flush operation returns success, although it didn't write anything.

Any ideas why this can happen? Did something changed in the latest version that can cause it?
I believe that in the past, when we used boards with exhausted flash cfstore returned error in the case.

Jenia

simonqhughes, 2016-10-31:
From: Jenia Kogan 
Sent: 28 October 2016 10:35
To: Rohit Grover
Cc: Nimrod Zimerman; Simon Hughes; Milosch Meriac
Subject: RE: Cfstore 

Forgot to mention that we are using synchronous mode

simonqhughes, 2016-10-31:
From: Rohit Grover 
Sent: 28 October 2016 11:19
To: Jenia Kogan
Cc: Nimrod Zimerman; Simon Hughes; Milosch Meriac
Subject: RE: Cfstore 

Jenia,

The K64F storage driver returns a special error code to indicate integrity failures. Refer to this (for the async. operation mode of the driver) or this (for sync. operation). The flash-journal, which is the current backend of the config-store, then translates such an error to another special error code, JOURNAL_STATUS_STORAGE_RUNTIME_OR_INTEGRITY_FAILURE, during the flush operation.

I don’t find any particular handling of this error code in configuration_store.c, particularly around the use of the log() operation. The configuration store uses a map to translate flash-journal errors to its own errors, and I don’t find a translation for JOURNAL_STATUS_STORAGE_RUNTIME_OR_INTEGRITY_FAILURE, which tells me that the config-store doesn’t yet handle integrity failures. Nevertheless, the config-store would attempt to return a default error code of DRIVER_ERROR_INTERNAL upon encountering an unknown failure from the underlying journal. I believe this should still be returned to the caller in the case of a failed flash.

You believe that you’re dealing with a failed flash storage. Perhaps you could add a few printfs in the failure path to verify that integrity failures are indeed occurring.

We could also wait for Simon Hughes to return on Monday and look into this more carefully than my cursory analysis.

rohit

simonqhughes, 2016-10-31:
Hi

Thanks for reporting the problem. Please can you help with the following questions?

>>we have board with exhausted flash. Write operations to flash are failed on this board.
What do you mean by "exhausted flash"? Do you mean that you're trying to store more data than memory? or than is available on the flash (512kB?)?

>> What is weird that we don't receive any indication about this failure from the cfstore driver.
If there was no error indication, how do you determine there to be an error? What operations are failing?

Please can you collect some debug logs in the same way as discussed when investigating previous issues, and append them to this ticket?

Thanks
Simon

TaniaMirzin, 2016-11-23:
@simonqhughes 
@jenia81 
Hi Simon,
As I know this problem was detected when device was tested with DAPLink 241, when the device was updated to DAPLink 242, the bug wasn't reproduced.
1. No, we are not trying to store more data than memory or flash storage.
 By "exhausted flash" we mean malfunction flash, as this bug reproduced only with specific board.
2. We trying to run flow that creates and write to flash enrollment keys. It seems that the keys was wrote to the flush, but in the next step when client tries to read these keys it is not there.

Please see the attached log.
We added some prints to indicate the problem,
You can see in line 394  
09:35:43 11:35:43.655 | D1 <-- DutThread: [20012d04]pv_service.c:374:PvServiceIotFinalize: Enter PvServiceIotFinalize, rot status 1, cert status 1
checks of rot keys and cert keys and at this point we see that both of them was wrote to the flush.
And later in line 461 
09:35:44 11:35:44.083 | D1 <-- DutThread: [20012bc4]pv_service.c:374:PvServiceIotFinalize: Enter PvServiceIotFinalize, rot status 0, cert status 1
We see that rot keys is not in flash.
It is important to mention that the bug reproduced about 30% of times it runs.
[CFSTOREֹFAILED.TXT](https://github.com/ARMmbed/mbed-os/files/608971/CFSTORE.FAILED.TXT)

TaniaMirzin, 2016-11-24:
@simonqhughes 
Hi Simon,
i have your old instruction of adding logs to cfstore:
Please collect configuration-store trace log for the test case you describe. To do this do the following:
•	edit cfstore_debug.h and change:
#define noCFSTORE_DEBUG
to
#define CFSTORE_DEBUG
edit configuration-store.c, changing this:
#ifdef CFSTORE_DEBUG
uint32_t cfstore_optDebug_g = 1;
uint32_t cfstore_optLogLevel_g = CFSTORE_LOG_NONE; /*CFSTORE_LOG_NONE|CFSTORE_LOG_ERR|CFSTORE_LOG_DEBUG|CFSTORE_LOG_FENTRY */
uint32_t cfstore_optLogTracepoint_g = CFSTORE_TP_NONE; /*CFSTORE_TP_NONE|CFSTORE_TP_CLOSE|CFSTORE_TP_CREATE|CFSTORE_TP_DELETE|CFSTORE_TP_FILE|CFSTORE_TP_FIND|CFSTORE_TP_FLUSH|CFSTORE_TP_INIT|CFSTORE_TP_OPEN|CFSTORE_TP_READ|CFSTORE_TP_WRITE|CFSTORE_TP_VERBOSE1|CFSTORE_TP_VERBOSE2|CFSTORE_TP_VERBOSE3|CFSTORE_TP_FENTRY */
#endif

to
#ifdef CFSTORE_DEBUG
uint32_t cfstore_optDebug_g = 1;
uint32_t cfstore_optLogLevel_g = CFSTORE_LOG_FENTRY; /*CFSTORE_LOG_NONE|CFSTORE_LOG_ERR|CFSTORE_LOG_DEBUG|CFSTORE_LOG_FENTRY */
uint32_t cfstore_optLogTracepoint_g = CFSTORE_TP_CLOSE|CFSTORE_TP_CREATE|CFSTORE_TP_DELETE|CFSTORE_TP_FILE|CFSTORE_TP_FIND|CFSTORE_TP_FLUSH|CFSTORE_TP_INIT|CFSTORE_TP_OPEN|CFSTORE_TP_READ|CFSTORE_TP_WRITE|CFSTORE_TP_VERBOSE1|CFSTORE_TP_VERBOSE2|CFSTORE_TP_VERBOSE3|CFSTORE_TP_FENTRY */
#endif

Can you check please if it still relevant? I see that the code has changed since this instruction?
Thank you.

jenia81, 2016-12-04:
@simonqhughes Any news about that?

jupe, 2016-12-05:
> As I know this problem was detected when device was tested with DAPLink 241, when the device was updated to DAPLink 242, the bug wasn't reproduced.

This sounds to me that root cause related somehow to software, not flash memory which burned out. But now we saw similar issue also with DAPLink 242 so root cause is still little mysterious.

jupe, 2016-12-05:
What I would do is to reduce all possible culprits, one by one. Maybe first thing is to verify that flash memory really works by creating memory health check solution. Maybe easiest way to do that is to create functionality to daplink that do verification during programming (https://github.com/mbedmicro/DAPLink/issues/182). That would validate every single flash operation straight way without affecting to current work flow. Then next simple step is to create dummy binary which uses entire flash memory and if we are able to flash that it should indicate that memory is healthy... 

This at least would tell if root cause is flash memory or not. If not, then perhaps there is really bug in software somewhere.

What do you think?

simonqhughes, 2016-12-05:
Hi

Here are some notes on the problem:
- I understand from the above that you suspect a CFSTORE flash related problem has been found in testing.
- The log file CFSTOREֹFAILED.TXT doesn't have any trace relating to CFSTORE so it doesn't provide any information (to me) as to whether CFSTORE is involved in this problem or not.
- The instructions posted above by Tania Mirzim for enabling CFSTORE trace are still correct.
- Please could you state on this ticket a brief description of the failure? e.g. what was the sequence of operations used to generate  CFSTOREֹFAILED.TXT?
- Is this problem isolated to one board? Or has it been observed on more than one board?
- You can use the flashjournal basicAPI.cpp test to test the integrity of the flash. If you run this 10 times and you see it fail 30% of the time, then this probably indicates a flash problem. 
- Please could you run the flashjournal greentea basicAPI test on the failing board 10 times and collect the output results? Please attach the results file to this ticket.
- If the flash journal tests work without a problem then please could you run the CFSTORE greentea tests on the failing board 10 times and collect the output results? Please attach the results file to this ticket.
- If you need help installing/setting up the greentea tea suite then get in touch via Skype/email.
- If the CFSTORE greentea tests pass fine, then please enable CFSTORE trace in the application and collect a log when then error is observed. Please attach the log to this ticket.

Best
Simon

TaniaMirzin, 2016-12-05:
@simonqhughes 
@jenia81

Hi Simon,
I tried to use your instruction for mbed-os 5.2.3 d5de476f74dd4de27012eb74ede078f6330dfc3f.
But i saw a few compilation errors.
I've added to sfctore_debug.h include of stdio.h ( i am not sure if it is ok, please refer to it).
It solved most of errors but i have still this:
/mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_svm.cpp:47:75: error: 'operation' was not declared in this scope
     CFSTORE_FENTRYLOG("%s: operation %d with status %d" , __func__, (int) operation, (int) status);
                                                                           ^
/mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_debug.h:74:22: note: in definition of macro 'CFSTORE_FENTRYLOG'
         printf(_fmt, __VA_ARGS__);                      \

Could you please provide us with relevant instructions?
Thank you.

simonqhughes, 2016-12-05:
Hi,

Apologies for the code breaking with debugging enabled. Here are the fixes.

The error logged above is reported on lin 47 of cfstore_svm.cpp. When I built mbed-os 5.2.3  d5de476, I found its was line 52 in error. Either way, comment out this line:
```
static void cfstore_svm_volume_manager_initialize_callback(int32_t status)
{
    //CFSTORE_FENTRYLOG("%s: operation %d with status %d" , __func__, (int) operation, (int) status);
}
```

There are also 2 errors in create.cpp test case: comment out the following lines:

```
    for(i = 0; i < max_num_kvs_create; i++)
    {
        memset(value_buf, 0, max_value_buf_size);
        snprintf(kv_name_tag_buf, kv_name_tag_len+1, "%0d", (int) i);
        ret = cfstore_create_kv_create(kv_name_min_len, kv_name_tag_buf, value_buf, kv_value_min_len/64 * (i+1));
        if(ret == ARM_CFSTORE_DRIVER_ERROR_OUT_OF_MEMORY){
            // comment out this line: CFSTORE_ERRLOG("Out of memory on %d-th KV, trying to allocate memory totalling %d.\n", (int) i);
            break;
        }
        /* revert CFSTORE_LOG for more trace */
        // comment out this line: CFSTORE_DBGLOG("Successfully stored %d-th KV bytes,  totalling %d.\n", (int) i, (int) bytes_stored);
    }

```

Then the application will build without errors.

Best
Simon

TaniaMirzin, 2016-12-06:
@simonqhughes 
@jenia81 

Hi Simon,
Thank you for the update.
I still have compilation problems.
1. Could you please refer to my previous fix of include "stdio.h".Is it ok?
2. i get this error now:
mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/configuration_store.c:59:44: error: 'ARM_Driver_Storage_MTD_K64F' undeclared here (not in a function)
 ARM_DRIVER_STORAGE *cfstore_storage_drv = &ARM_Driver_Storage_MTD_K64F;
Please help us to resolve it.
BR,
TaniaMirzin.

simonqhughes, 2016-12-08:
Hi

Please provide the mbed command line that you're using to build with?

Thanks
Simon

TaniaMirzin, 2016-12-08:
@simonqhughes 
@jenia81 

Hi Simon please check this:
python -u mbed-os/tools/build.py -t GCC_ARM -m K64F --profile mbed-os/tools/profiles/debug.json --source mbed-os --build .build --artifact-name libmbed-os -v -DMBEDTLS_CMAC_C -DMBEDTLS_PEM_WRITE_C -DMBEDTLS_CTR_DRBG_MAX_REQUEST=2048 -DSTORAGE_DRIVER_CONFIG_HARDWARE_MTD_ASYNC_OPS=0 -DSTORAGE_CONFIG_HARDWARE_MTD_K64F_ASYNC_OPS=0" in mbed-os

simonqhughes, 2016-12-08:
Hi

```
mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/configuration_store.c:59:44: error: 'ARM_Driver_Storage_MTD_K64F' undeclared here (not in a function)
 ARM_DRIVER_STORAGE *cfstore_storage_drv = &ARM_Driver_Storage_MTD_K64F;
 Please help us to resolve it.
```

To get the above error then CFSTORE_CONFIG_BACKEND_FLASH_ENABLED must not be defined. Why is this happening?

Simon

simonqhughes, 2016-12-14:
I have submitted pull request to fix the debug issues here:

https://github.com/ARMmbed/mbed-os/pull/3368

simonqhughes, 2016-12-14:
From: Jenia Kogan 
Sent: 14 December 2016 12:06
To: Simon Hughes
Cc: Tania Mirzin
Subject: RE: CFSTORE issue 3167

Hi Simon,

Tania is OOO till Sunday.
She'll look on this on Sunday

Jenia

From: Simon Hughes 
Sent: Wednesday, December 14, 2016 1:55 PM
To: Tania Mirzin
Cc: Jenia Kogan
Subject: CFSTORE issue 3167

Hi

Please could you let me know whether you’re now unblocked with the following issue?

https://github.com/ARMmbed/mbed-os/issues/3167

Can it be resolved?

Thanks
Simon

TaniaMirzin, 2016-12-18:
@simonqhughes 
@jenia81 
Hi Simon,
I am using now your fix version ad8a77d1aeef41ae1bc40096cbce85921f846a90.
The problems of include headers and log function prototype are fixed.

But I still have the same problem with ARM_Driver_Storage_MTD_K64F error:
/mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/configuration_store.c:59:44: error: 'ARM_Driver_Storage_MTD_K64F' undeclared here (not in a function)
 ARM_DRIVER_STORAGE *cfstore_storage_drv = &ARM_Driver_Storage_MTD_K64F;

You wrote this " To get the above error then CFSTORE_CONFIG_BACKEND_FLASH_ENABLED must not be defined. Why is this happening?"
So I checked CFSTORE_CONFIG_BACKEND_FLASH_ENABLED and it is defined :
mbed-os/features/storage/FEATURE_STORAGE/cfstore/source/cfstore_config.h:56:6: error: stray '#' in program
  err #define CFSTORE_CONFIG_BACKEND_FLASH_ENABLED
      ^
I've added "err" to cfstore_config.h line 56 to check it, and as you can see it is defined as I reach this code line.

Could you please check again that I am performing the right steps for enabling of sfctore logs:
1. in sfctore__debug.h : change #define noCFSTORE_DEBUG  to #define CFSTORE_DEBUG.
2.  in configuration-store.c

Change this:
#ifdef CFSTORE_DEBUG
uint32_t cfstore_optDebug_g = 1;
uint32_t cfstore_optLogLevel_g = CFSTORE_LOG_NONE; /*CFSTORE_LOG_NONE|CFSTORE_LOG_ERR|CFSTORE_LOG_DEBUG|CFSTORE_LOG_FENTRY */
uint32_t cfstore_optLogTracepoint_g = CFSTORE_TP_NONE; /*CFSTORE_TP_NONE|CFSTORE_TP_CLOSE|CFSTORE_TP_CREATE|CFSTORE_TP_DELETE|CFSTORE_TP_FILE|CFSTORE_TP_FIND|CFSTORE_TP_FLUSH|CFSTORE_TP_INIT|CFSTORE_TP_OPEN|CFSTORE_TP_READ|CFSTORE_TP_WRITE|CFSTORE_TP_VERBOSE1|CFSTORE_TP_VERBOSE2|CFSTORE_TP_VERBOSE3|CFSTORE_TP_FENTRY; */
#endif

To :
#ifdef CFSTORE_DEBUG
uint32_t cfstore_optDebug_g = 1;
uint32_t cfstore_optLogLevel_g = CFSTORE_LOG_FENTRY; /*CFSTORE_LOG_NONE|CFSTORE_LOG_ERR|CFSTORE_LOG_DEBUG|CFSTORE_LOG_FENTRY */
uint32_t cfstore_optLogTracepoint_g = CFSTORE_TP_CLOSE | CFSTORE_TP_CREATE | CFSTORE_TP_DELETE | CFSTORE_TP_FILE | CFSTORE_TP_FIND | CFSTORE_TP_FLUSH | CFSTORE_TP_INIT | CFSTORE_TP_OPEN | CFSTORE_TP_READ | CFSTORE_TP_WRITE | CFSTORE_TP_VERBOSE1 | CFSTORE_TP_VERBOSE2 | CFSTORE_TP_VERBOSE3 | CFSTORE_TP_FENTRY
#endif
Thank you.

sg-, 2017-02-03:
@simonqhughes Can this be closed?

closed 2017-10-27 by ChiefBureaucraticOfficer